3drobotics/solodevguide #123: How to keep connected to internet on the development host while connected to Solo

mrpollo, 2015-09-11:
**Author:** Kelly Schrock 
**Source:** [DIYD](http://diydrones.com/profiles/blogs/official-development-guides-for-3dr-solo?commentId=705844%3AComment%3A2090644&xg_source=msg_com_blogpost)

```
This is great. I have a dumb question though... I have no problem switching to the Solo's wifi and logging in, but what's a good way to get access to both the Solo and an internet connection at the same time? 
```

mrpollo, 2015-09-11:
Use a second network adapter eg: `Ethernet`

mrpollo, 2015-09-11:
@hamishwillee  this is getting asked frequently, perhaps add to guide?

closed 2015-09-11 by mrpollo

hamishwillee, 2015-09-14:
@mrpollo I don't understand the question or the answer. Probably because I don't have a solo. 

My understanding is that you SSH into solo. YOu can then set up a tunnel so solo can access the Internet. So aren't you therefore "In SOLO" and accessing the Internet at the same time? 

How would you use a second ethernet adapter - does Solo have a cable ethernet port?

Happy to help, but need to really understand the problem and the solution.

kellyschrock, 2015-09-14:
Turns out, this is trivially easy. Hook an Ethernet cable to your laptop, and use wifi to connect to Solo.

hamishwillee, 2015-09-14:
@kellyschrock Still lost - [this doesn't answer my comment](https://github.com/3drobotics/solodevguide/issues/123#issuecomment-139932530).

kellyschrock, 2015-09-14:
Ah, OK. Sorry... 

Plugging an Ethernet cable into your laptop gives it (the laptop) internet access, while still allowing you to join the Solo's wi-fi network and connect to it at the same time. Once you've gone through the guide and installed solo-utils (including following through the section on SSH setup), you have a command you can run from an SSH session while you're logged into the Solo:

solo-utils tunnel-start

This takes a few seconds, and when complete, creates a tunnel that allows Solo to use your laptop's internet access (provided by the Ethernet cable) to download further software packages that can be installed on the Solo. So in that case, you are "In Solo" and accessing the internet at the same time.

hamishwillee, 2015-09-14:
@kellyschrock It never occurred to me that you can only connect to one wifi network at a time from a single adapter - so if you connect to Solo network then you can't be connected to the Internet. Now I'm feeling dumb! I'll fix the docs.

kellyschrock, 2015-09-14:
Great! I should have known it was as easy as plugging in an Ethernet cable.
But to do that, I had to get up, walk around the back of a different
computer, pull a plug out, plug it in somewhere else, etc. That's a lot of
work! Better to ask what to do on a forum. :-)

hamishwillee, 2015-09-14:
Cool! @mrpollo This is addressed by my [patch](https://github.com/3drobotics/solodevguide/pull/120) (this [changelist](https://github.com/3drobotics/solodevguide/commit/548dc0cd1af5f96eddd9d7154c3f127c9f544999) ) which adds a bullet:

> Connect the host computer to the Internet using an Ethernet cable or other spare network adapter (your wifi adapter will be attached to the Solo network!)

OXINARF, 2015-09-14:
What would be cool (and possible) was that solo-utils had an option so that the Solo's controller would connect to the user/developer's Wifi network and act like a true router. That way the Ethernet cable wouldn't be needed.

hamishwillee, 2015-09-14:
Thanks @OXINARF. Agree that would be cool. Raised separate isssue #132 for discussion.

tcr3dr, 2015-09-25:
@OXINARF, @kellyschrock, we recently rewrote the guide to use a different and more robust method of connecting to the Internet. Check out http://dev.3dr.com/starting-utils.html for updated instructions and feel free to continue this discussion if you run into issues. Hopefuly this makes the development process simpler!