3drobotics/solodevguide #125: Configure tools requires `tunnel-start`

Connecting to the internet should come before. 

Also anytime steps switch between host and solo for running terminal commands a note should be made(a note was made for switching to host but never for switching back to solo). I just tried to run solo-utils on my host machine because I was never instructed to run it on solo. What may be a face palm for me, may be pain point for beginner.

tcr3dr, 2015-09-11:
**Author:** Daniel Nugent
**Page:** http://dev.3dr.com/

Configure tools requires `tunnel-start`

Connecting to the internet should come before. 

Also anytime steps switch between host and solo for running terminal commands a note should be made(a note was made for switching to host but never for switching back to solo). I just tried to run solo-utils on my host machine because I was never instructed to run it on solo. What may be a face palm for me, may be pain point for beginner.

djnugent, 2015-09-11:
Man does the entire message really need to be the subject of the issue haha. Also it didn't link the page properly. I was on starting-utils

djnugent, 2015-09-11:
feedback tool--

djnugent, 2015-09-11:
Dead link in the "tip" on internet connection stuff

hamishwillee, 2015-09-11:
@djnugent Thanks for that. I believe my patch of yesterday (now) fixes this (I fixed up the tip, and put "connect to internet" first).
https://github.com/3drobotics/solodevguide/pull/120

tcr3dr, 2015-09-25:
I believe this is more adequately addressed by the docs, and clarifying the host/solo distinction is #142 and #143.

closed 2015-09-25 by tcr3dr