3drobotics/solodevguide #128: Smart update needs to be called before calling resize-fs

hamishwillee, 2015-09-14:
Author: Kelly Schrock 
Source: [DIYD](http://diydrones.com/profiles/blogs/official-development-guides-for-3dr-solo?commentId=705844%3AComment%3A2090644&xg_source=msg_com_blogpost)

> OK, next error. I'm trying to run solo-utils resize-fs, and getting this error:
error: missing packages. please run:
  solo-utils tunnel-start
  smart install parted libss2 e2fsprogs e2fsprogs-badblocks e2fsprogs-mke2fs lsof
The last command is failing. If I run smart install on individual packages, I get a short progress indicator ("####", etc) and an error (e.g. "error: 'parted' matches no packages"). Tunnel is running, so I'm not sure what the deal is.
Thanks

> OK, just realized I need to run "smart update" before all of this. Might want to update the documentation with that, but people might already know to do that (I didn't, just guessed). Going quite well otherwise.

So the requirement here is to properly document the smart update

hamishwillee, 2015-09-14:
@mrpollo My proposal is to 

1. duplicate the [Installing Packages](http://dev.3dr.com/starting-installing.html#installing-packages) section (add the `smart update` line) to "connecting to the Internet" [in my new patch](https://github.com/3drobotics/solodevguide/blob/hamishwillee-patch-1/book/starting-utils.md#connecting-to-the-internet).  . 

   ```
solo-utils tunnel-start
smart update
   ```

   This section comes before resizing the partition and is *likely* to be done first. 

1. Add a note to the  [running the utils section](https://github.com/3drobotics/solodevguide/blob/hamishwillee-patch-1/book/starting-utils.md#running-solo-utils) that basically says "you should connect to the Internet and call `smart update` before configuring other tools and resizing the partition"

Does that sound reasonable?

tcr3dr, 2015-09-16:
@hamishwillee That makes sense. Also, lastly, the tool should specify updating `smart update`. 

https://github.com/3drobotics/solodevguide/blob/master/tools/solo-utils/solo-utils-resize-fs#L14

If you added that line, I'd publish a new solo-utils version.

hamishwillee, 2015-09-16:
@tcr3dr Great. Patch for the docs is here https://github.com/3drobotics/solodevguide/pull/144

tcr3dr, 2015-09-16:
Awesome.

As of https://github.com/3drobotics/solodevguide/commit/4a1d6e54cd2e731335f86e585bce98dd3d8774a7 we will not need to run this twice, and as of https://github.com/3drobotics/solodevguide/commit/8c7994aead6cf09f64df01101565b3894e6cb24a we won't need to run `smart install` at all before the resize-fs step. I don't think this impacts the logs, as the warning is only shown if that installation failed on tools setup. So I believe this is resolved now.

tcr3dr, 2015-09-16:
Ha, just kidding, this isn't fixed. Still investigating.

hamishwillee, 2015-09-16:
@tcr3dr Thanks. When this is all "finished" let me know the precise changes required (ie if there is an additional argument for size of the partition, whether it has a default, how it is specified, what we recommend, whether or not we still need to advise that this could fail, and if so to just run it again.

tcr3dr, 2015-09-17:
Will do. We'll keep the advice to "run it again". It just shouldn't fail the first time.

tcr3dr, 2015-09-26:
This actually is integrated into the `solo resize` step now, so it shouldn't be an issue for Solodevguide.

closed 2015-09-26 by tcr3dr