3drobotics/solodevguide #129: resize-fs might fail / need to be called again.

hamishwillee, 2015-09-14:
Author: Kelly Schrock 
Source: [DIYD](http://diydrones.com/profiles/blogs/official-development-guides-for-3dr-solo?commentId=705844%3AComment%3A2090644&xg_source=msg_com_blogpost)

> For those of you updating whatever documentation you're updating, you might mention that solo-utils resize-fs might fail, and you'll need to go at it again. I tried it once, and it barfed a few error messages. After a reboot, everything seemed OK, but the partition wasn't big enough to accommodate a git installation. So I ran it again, and it "took".

hamishwillee, 2015-09-14:
@mrpollo The last section of the docs related to this doesn't make much sense to me:

> To expand the root partition from 90Mb to 250Mb, you will delete and recreate your /log directory. Ensure you have any important data backed up. Then run:

> `solo-utils resize-fs`

1. Given that there are no parameters to `resize-fs` does this mean that you can only expand to 250Mb?
   
   * Is this reversible
   * FMI, if this is something you might want to do, why isn't it the "standard"
   * Why expand the root partition?

1. The "you will delete and recreate your /log directory" seems separate. I THINK that we are trying to say (please confirm):

   > "Note: Changing the root partitions size will delete and recreate your /log directory. Ensure you have any important data backed up.

    > To expand the root partition from 90Mb to 250Mb run:

    > `solo-utils resize-fs`

I can then add Kelly's note above as a tip at the end.

tcr3dr, 2015-09-15:
We need to explore the "how much space is enough space" question.

hamishwillee, 2015-09-15:
@tcr3dr True. I can fix this "for now" though if you can answer https://github.com/3drobotics/solodevguide/issues/129#issuecomment-139935863

tcr3dr, 2015-09-16:
> Given that there are no parameters to resize-fs does this mean that you can only expand to 250Mb?

Yup. We can easily change this to accept a parameter.

> Is this reversible

Not easily, except for a factory reset. You could probably shrink the root partition with other Linux tools, but the ones SDG installs only expands the root parition.

> FMI, if this is something you might want to do, why isn't it the "standard"

Solo ships with as much log space as possible and as little extra space on the root partition as it needs. If you install more pacakges on Solo, you are thus using more capabilities than stock Solo required.

> Why expand the root partition?

To allow you to install more software, thus performing various kinds of examples found in SDG.

> The "you will delete and recreate your /log directory" seems separate. I THINK that we are trying to say (please confirm): "Note: Changing the root partitions size will delete and recreate your /log [...]

Yup, deleting and recreating /log is fully taken care of by the tool.

Also, it does fail and calling it again is the fix. I have to find out what the failure mode is, but still the way it works should generally allow you to re-run the tool to try again.

hamishwillee, 2015-09-16:
Thanks @tcr3dr. I think this is addressed properly by #145.

I don't know whether or not it makes sense to make the size configurable. I guess it is possible that some people might set up multiple space-consuming environments while others would still want a lot of log area? Either case, would make sense to retain the current default.

Maybe spawn a separate issue for that.

tcr3dr, 2015-09-25:
See https://github.com/3drobotics/solo-cli/issues/6 and https://github.com/3drobotics/solo-cli/issues/5 .

hamishwillee, 2015-09-28:
@tcr3dr I think we can close this - the actual issue here is now addressed, and the other discussions we have are covered by other issues.

closed 2015-09-28 by tcr3dr