3drobotics/solodevguide #130: I need to bypass arming checks. How?

tcr3dr, 2015-09-14:
How brave are you?

```
param set ARMING_CHECK 0
param set FS_THR_ENABLE 0
param set FS_GCS_ENABLE 0
param set EKF_CHECK_THRESH 0
```

hamishwillee, 2015-09-14:
Doesn't the first check disable everything needed for arming?

tcr3dr, 2015-09-14:
@hamishwillee This is correct. The other configurations are for various SITL options. I need to look into exactly what they do, but it will play a part in a forthcoming simulation or building topic.

hamishwillee, 2015-09-14:
@tcr3dr Ok - thanks. Some care needed when you add this. The original DK examples all did the param set programmatically, which is pretty bad from a copy and paste point of view. I changed the instructions so that disarming the checks happened only in the simulator setup, and was accompanied by warnings.

And of course those instructions above are MAVProxy commands, so we need to "de-GCS-ify" them.

tcr3dr, 2015-09-15:
@hamishwillee Precisely.

GCSification isn't too bad if we talk about using MAVProxy as a way to communicate with Solo. Not sure how that will work yet, but using DK is also an option.

hamishwillee, 2015-09-16:
@tcr3dr 

1. There is precedence for using MAVProxy as the developer GCS for both ArduPilot and DroneKit. I plan on adding a section to this for DKPY2 now that MAVProxy is no longer a requirement. 
2. My very strong preference is that this isn't a generic "how to bypass arming checks" topic but is part of "simulation" as you have indicated. 
3. > Is it weird for us to basically say "use python in interpreter mode"?
   
    Yes :-) That said, I think this is the easiest/lowest setup cost for user, and would be fine with the appropriate wording that introduces this as an option among many. For example:

   > Disable the arming checks on the Simulator by setting the ARMING_CHECK parameter to 0. You can do this in via a GCS or using DroneKit. 

   > WARNING: Disabling the arming check should only be done on the Simulator, and never hard coded into DroneKit-Python scripts.

   > For example to disable the arming check from the python interpreter:

   >   ```
$ python
>>> import dronekit
>>> v = dronekit.connect('udpout:10.1.1.10:14560')
<dronekit...>
>>> v.parameters['ARMING_CHECK'] = 0
   ``` 

   > To disable the arming check using MAVProxy:

   >   ```
param set ARMING_CHECK 0
   ``` 

  ... Then something about the other arming checks.

I don't thing the `udpout:` is needed in:
```
v = dronekit.connect('udpout:10.1.1.10:14560')
```

kawfee, 2015-11-02:
Finally a thread relevant to my question!  Due to current FAA rules, university people learning how how to fly our X8 3DR drones counts as commercial activity, so we have to do everything indoors in a mesh cage => no GPS. (Not wishing to start a discussion of FAA rules here.) 

I know that this is the "Solo" discussion area, but any words of wisdom on how to get started with X8/DroneKit? I know how to program just fine, but know very little about DroneKit and even less about using DroneKit indoors.

Thanks.

hamishwillee, 2015-11-03:
@kawfee For general dronekit information [read the documentation](http://python.dronekit.io/). There is no information on using it indoors because the autopilot has limited positioning indoors. There are some people who are discussing options on the [gitter channel](https://gitter.im/dronekit/dronekit-python) so recommend you raise any questions there.