3drobotics/solodevguide #133: On the section Optimizing SSH add a line stating that you need to create your ssh keys with ssh-keygen

tcr3dr, 2015-09-15:
**Author:** Jaime Machuca
**Page:** http://dev.3dr.com/starting-network.html

On the section Optimizing SSH add a line stating that you need to create your ssh keys with ssh-keygen

tcr3dr, 2015-09-15:
We should start the "Using SSH" section with something similar to [the Git book on using public SSH](https://git-scm.com/book/en/v2/Git-on-the-Server-Generating-Your-SSH-Public-Key#_generate_ssh_key) or the [Github guide](https://help.github.com/articles/generating-ssh-keys/).

Probably we can condense this into

> Do you have a .ssh directory in your home folder, or are you getting "permission denied" errors when connecting to solo? You may first need to generate an SSH key by running `ssh-keygen -t rsa -b 4096 -C "your_email@example.com"`. See [this guide](https://help.github.com/articles/generating-ssh-keys/) for more info.

tcr3dr, 2015-09-15:
@hamishwillee I actually don't know how you generate an SSH key on Windows; would Putty, for example, take care of this, or openssh?

hamishwillee, 2015-09-15:
@tcr3dr Putty comes with puttygen tool which most people use for this purpose. The tool is fairly intuitive, but there are some [reasonable instructions here](https://winscp.net/eng/docs/ui_puttygen).

You can also use OpenSSH but you need to install Cygwin first. A bit of a pain, and I always use Puttygen.

hamishwillee, 2015-09-15:
Re https://github.com/3drobotics/solodevguide/issues/133#issuecomment-140579256, you need to make it clear that this key is on the host. I also assume that you don't need to do anything Solo side?

Your note is good, but I'd change

> To SSH into Solo, ensure you have an SSH-capable machine that is connected to the Controller's private network. From your terminal, type:

To

>To SSH into Solo, ensure you have an SSH-capable machine that is connected to the Controller's private network. 

> [note aside] An SSH-capable machine is one that has SSH [link here to download location] installed and for which you have created an SSH key (without the key you may get  "permission denied" errors when connecting to solo). Instructions on how to create the key are provided here (Linux, MacOSX) and here (Windows) [link to puttygen]

> From your terminal, type:

(point being that what we're talking about here is "how to set up an SSH-capable machine".

If we want to spell out how to set up putty and puttygen, tne consider putting a separate document in the Appendix on this.

tcr3dr, 2015-09-16:
Probably not worth setting up putty and puttygen. I know there's a variant of openssh that works on Windows like [the one on chocolatey](https://chocolatey.org/packages/openssh), or there's cygwin, and between those some willing and competent developer may be able to use the relevant guides. Linking to puttygen sounds fine by me though.

tcr3dr, 2015-09-16:
Or maybe http://www.mls-software.com/opensshd.html ?

hamishwillee, 2015-09-16:
Second link looks promising. I'd link to both for Windows. If they are already set up they will just ignore this note anyway.

tcr3dr, 2015-09-25:
I added a link to PuTTY and Cygwin + OpenSSH, after some reflection, since I believe that's really the most accessible and updated tools.

Someone recommended msysgit + OpenSSH as a recommended option too, which I should experiment with.

hamishwillee, 2015-09-28:
@tcr3dr Good fix - having those two would cover most users.

We still need to say something about generating the keys right?