3drobotics/solodevguide #134: Its not clear where solo-utils live, it should be more clear where to run the solo-utils scripts. Make it clear you have to be ssh into solo and that these scripts need to run on solo. The installation of pip, needs to be done after the solo is connected to the internet using the tunnel. So I would change the order, of Configure Tools and Connecting to the Internet.

tcr3dr, 2015-09-15:
**Author:** Jaime Machuca
**Page:** http://dev.3dr.com/starting-utils.html

Its not clear where solo-utils live, it should be more clear where to run the solo-utils scripts. Make it clear you have to be ssh into solo and that these scripts need to run on solo. The installation of pip, needs to be done after the solo is connected to the internet using the tunnel. So I would change the order, of Configure Tools and Connecting to the Internet.

jmachuca77, 2015-09-15:
related to #125

hamishwillee, 2015-09-15:
I think all of these are addressed by https://github.com/3drobotics/solodevguide/pull/120 which is merged but not yet live. @jmachuca77  Can you check?

jmachuca77, 2015-09-15:
Yes, it seems they are. When I left feedback through the page feedback I did not realize this issues were already addressed. I think this can be closed.

hamishwillee, 2015-09-15:
@jmachuca77 - no way for you to know until it goes live. Keep those fixes coming - really appreciate the information.

closed 2015-09-15 by hamishwillee