3drobotics/solodevguide #142: Visual color indicator for code run on solo vs on PC.

tcr3dr, 2015-09-15:


hamishwillee, 2015-09-15:
I like this idea, but I'm not sure that there is a "colour" that implies Solo vs. PC. Perhaps an icon might work better?

My only minor concern with this is that during preview and on Github you can't see the icon - this is already a pain when viewing content in asides as you edit. Not expecting any action, just saying :-)

tcr3dr, 2015-09-16:
@hamishwillee I agree, the effect is good but it won't work on Github. Probably we'll have to think of something more clever, but we can get started now!

tcr3dr, 2015-09-16:
In the future we could use ```solo-sh and ```host-sh which would work on Github and on Gitbook. Not worth writing an extension for yet though.

hamishwillee, 2015-09-16:
@tcr3dr That  ```solo-sh is a **very** nice solution. Also worth considering for notes, tips, warnings etc.

If we want to use this now I don't see an easier/safer way to do this.

tcr3dr, 2015-09-16:
@hamishwillee Gonna take a bit. Basically I have to learn gitbook's plugin code, whereas #143 is effectively just a stylesheet fix.

hamishwillee, 2015-09-16:
@tcr3dr Good point. YOu could still add an icon using #143 of course using :before or similar.

tcr3dr, 2015-09-25:
Okay, moving ahead with https://github.com/3drobotics/solodevguide/pull/180 now.

tcr3dr, 2015-10-28:
Done.

closed 2015-10-28 by tcr3dr