3drobotics/solodevguide #156: solo-utils tunnel-start

vasnakh, 2015-09-17:
when i start this command it asks for host user-name and it automatically selects host 10.1.1.194:22 and after typing host username it gives ssh: connect to host 10.1.1.149 port 22: Connection timed out
failed. Could you please help me on how to fix this?

vitorguizilini, 2015-09-19:
I have a very similar issue. After running "solo-utils tunnel start" and typing host username, I receive:

"ssh: connect to host 10.1.1.174 port 422: Connection refused"

Solo-utils seems to have been installed propertly, and I am sure my laptop is also connect to the internet via Ethernet cable.  The key was also added to the authorized list.

hamishwillee, 2015-09-22:
@tcr3dr Did you see this one?

tcr3dr, 2015-09-22:
Yup. We'll be replacing this machinery very shortly, so hold on tight.

tcr3dr, 2015-09-25:
Hi @vasnakh, @vitorguizilini,

We recently rewrote the guide to use a different method of connecting to the Internet, which should be much more successful.

* Please try reading through this page and following the instructions: http://dev.3dr.com/starting-utils.html
* In particular, this section has changed: http://dev.3dr.com/starting-utils.html#connecting-to-solo-and-the-internet

We hope that this new mechanism of connecting to the Internet will be easier and allow you to proceed with the guide. Feel free to update this thread with any issues or questions you have.

tcr3dr, 2015-09-26:
One more point, we no longer require you to have SSH enabled on your PC for this to work. Solo CLI should run out-of-box across all platforms.

vasnakh, 2015-09-26:
That's great, but when executing solo wifi command with network name and password it says that the controller firmware version needs to be 1.1.16 or newer but as I tried to update it through my ipad and android phone the latest update was 1.1.15! Would you let me know how I can get 1.1.16 firmware for the controller therefor resulting in solo wifi command to work? I appreciate your help, thank you.

vasnakh, 2015-09-26:
Hi @tcr3dr 

That's great, and thank you. I went through the steps again but when executing solo wifi command with my network name and password it says that the controller firmware version needs to be 1.1.16 or newer and as I tried to update it through my ipad and android phone the latest update was 1.1.15! Would you let me know how I can get 1.1.16 firmware for the controller therefore resulting in solo wifi command to work? I appreciate your help, thank you.

vitorguizilini, 2015-09-28:
Yeah, I am having the same issue with the controller version. It requires 1.1.16, but mine is 1.1.15. I tried running "solo update controller latest" to upgrade, but i gives me the following log:

you are about to update the Controller.
this preserves all your local changes to Solo, but compatibility
with newer updates is not guaranteed.
proceed to perform update? [y/N] y
waiting for Internet connectivity...
Traceback (most recent call last):
  File "/usr/local/bin/solo", line 9, in <module>
    load_entry_point('soloutils==1.0.0', 'console_scripts', 'solo')()
  File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 351, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 2363, in load_entry_point
    return ep.load()
  File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 2088, in load
    entry = __import__(self.module_name, globals(),globals(), ['__name__'])
  File "/usr/local/lib/python2.7/dist-packages/soloutils/__main__.py", line 32, in <module>
    soloutils.update.main(args)
  File "/usr/local/lib/python2.7/dist-packages/soloutils/update.py", line 118, in main
    updates = releases(updatepath)
  File "/usr/local/lib/python2.7/dist-packages/soloutils/update.py", line 24, in releases
    root = urlparse.urljoin(urlparse.urljoin(SERVERADDR, where), 'update/')
NameError: global name 'SERVERADDR' is not defined

Any ideas on how I could go past that?

tcr3dr, 2015-09-28:
Hi @vasnakh, @vitorguizilini, we're working on fixing the update command and will have this out soon.

In the meantime, I tested against `1.1.15` and realize that the minimum version was incorrect. Feel free to reinstall Solo CLI and try with your current version:

```
pip install -UI git+https://github.com/3drobotics/solo-cli
```

vitorguizilini, 2015-09-29:
Great, thanks. I reinstalled Solo CLI and there was no version conflict.  However, now I get the following log:

connecting to the Controller...
restarting DNS forwarder and DHCP server: dnsmasq... 
stopping DNS forwarder and DHCP server: dnsmasq... stopped /usr/bin/dnsmasq (pid 771)
done.
starting DNS forwarder and DHCP server: dnsmasq... done.
done.
Successfully initialized wpa_supplicant
rfkill: Cannot open RFKILL control device
udhcpc (v1.21.1) started
Sending discover...
Sending discover...
Sending discover...

And Solo keeps making some beeps and blinking the LEDs once every 2-3 seconds. 

Sorry to insist on this matter, but I would like to get this sorted out as quickly as possible to resume working on software development.

tcr3dr, 2015-09-30:
Hi @vitorguizilini,

There are two possible culprits. The first is that I recommend you restart your Controller—repeated use of the `wifi` command can leave it hanging, and it requires a reset to work again.

The second is incorrect WiFi credentials. That will not give you a "ssid/password incorrect" error, instead it will just continue to write out "Sending discover..."

If after restarting the Controller once or twice and verifying the correct credentials does not work, update me here and I'll respond with what diagnostics would best address this.

vasnakh, 2015-09-30:
Hi @tcr3dr ,

I have actually tried to run the same command as well, but I exactly get the same message that @vitorguizilini gets. I tried resetting the controller and also made sure my user/pass is correct but still I get the same error!

tcr3dr, 2015-09-30:
Thanks for your feedback (and patience!) In light of that, I will go ahead and try to replicate the problem, starting by testing on a larger variety of WiFi networks. Hopefully I can recreate the issue and get back to you both.

In the meantime, if you have time to kill, try to connect to a very different WiFi network (e.g. a phone hotspot, neighbor's wifi, etc.). My first hunch is that perhaps the issue varies by 2.4Ghz/5Ghz or 802.11 version.

tcr3dr, 2015-10-02:
Just an update, I've identified the problem and we'll be shipping a fix tomorrow.

vasnakh, 2015-10-02:
@tcr3dr; Sounds great! I will look forward to that. Please let me know when the fix is uploaded.

tcr3dr, 2015-10-02:
Hi @vasnakh, please try running this:

```
sudo pip install -UI git+https://github.com/3drobotics/solo-cli@tcr-updates
```

This should fix `solo wifi` and also enable `solo update controller 1.1.15` (if necessary). We'll merge this to master soon.

vasnakh, 2015-10-03:
@tcr3dr 

I tried to run it but it gives me the error below:

Could not find a tag or branch 'tcr-updates', assuming commit.
error: pathspec 'tcr-updates' did not match any file(s) known to git.
  Complete output from command git checkout -q tcr-updates:

I guess the path is wrong since as I click on the link it says not found!

vasnakh, 2015-10-03:
@tcr3dr 

I actually found out that the correct command should have been sudo pip install -UI git+https://github.com/3drobotics/solo-cli@tcr-updates. It worked and it installed correctly but as I run solo wifi it just says connecting to controller....... and nothing else comes up at all. It just stops right here.

tcr3dr, 2015-10-03:
@vasnakh Sorry for the incorrect link—glad you got it working.

I believe the issue is just a weird network issue. Try rebooting the Controller. Then reconnect to the SoloLink_XXX WiFi network. If you continue to get issues, try running `ssh root@10.1.1.1` from your command line; this should prompt you for a passphrase to log into your Controller.

If this fails, let me know and I'll come up with a better diagnostic.

vasnakh, 2015-10-03:
@tcr3dr 

Just tried it and after doing so it asked for password. After running that I tried solo wifi again and this time it came back to the first issue which is shown below:

connecting to the Controller...
restarting DNS forwarder and DHCP server: dnsmasq... 
stopping DNS forwarder and DHCP server: dnsmasq... stopped /usr/bin/dnsmasq (pid 771)
done.
starting DNS forwarder and DHCP server: dnsmasq... done.
done.
Successfully initialized wpa_supplicant
rfkill: Cannot open RFKILL control device
udhcpc (v1.21.1) started
Sending discover...
Sending discover...
Sending discover...

and that keeps running as before

vitorguizilini, 2015-10-07:
Hi,

Sorry for the delay in replying, some other stuff came up... The trc-updates link is not available anymore, so I tried again with the original one. Still the same issue (exactly the same as the reply above me).

vasnakh, 2015-10-09:
@tcr3dr; Have you been able to find any solutions yet? Please let me know here if there is any update. Thank you.

vitorguizilini, 2015-10-12:
I am also waiting for a solution. Let us know if there are any updates available. Thank you.

robinabraham, 2015-10-15:
I am running into the same issue as well. Tried connecting over multiple networks, both 2.4Ghz and 5Ghz. Same result. Keeps repeating the "Sending discover..." message, and Solo keeps beeping. Thanks.

vitorguizilini, 2015-10-27:
Hi,

I'm just wondering if there are any developments, because I still haven't been able to move past this "Sending discover..." message.  Thanks.

OXINARF, 2015-10-27:
@vitorguizilini Do you have the latest version of the solo tool? There was an update yesterday.

tcr3dr, 2015-10-28:
@vasnakh, @vitorguizilini, @robinabraham,

As @OXINARF mentioned, solo-cli has been updated with a new method for running `solo wifi`. The old method of the command did not work with differing wifi channels, which caused the errors you've seen. We've rewritten it to connect to your local network first, then re-establish connection with Solo. It should address a lot of these issues.

Install from [the solo-cli repo](http://github.com/3drobotics/solo-cli) or just run:

```
sudo pip install -UI git+https://github.com/3drobotics/solo-cli
```

Give it a try and feel free to continue this thread if you see issues.

vitorguizilini, 2015-10-29:
Great, this new update seems to work perfectly, thanks!

tcr3dr, 2015-10-29:
@vitorguizilini :)

robinabraham, 2015-10-29:
Hi,

The update works and allows the controller to also connect to my local Wifi network. However, the drone is still unable to connect to Wifi. So, when I try other commands like "solo install-smart", I get errors (pasted below). 

Fetching information for 'cortexa9hf_vfp_neon_mx6'...                          
-> http://solo-packages.s3-website-us-east-1.amazonaws.com/.../repomd.xml      
repomd.xml                                                               [  6%]
error: Failed acquiring release file for 'cortexa9hf_vfp_neon_mx6':
error: http://solo-packages.s3-website-us-east-1.amazonaws.com/3.10.17-rt12/cortexa9hf_vfp_neon_mx6/repodata/repomd.xml: [Errno -2] Name or service not known
                                                                               
Fetching information for 'all'...
-> http://solo-packages.s3-website-us-east-1.amazonaws.com/.../repomd.xml      
repomd.xml                                                               [ 31%]
error: Failed acquiring release file for 'all':
error: http://solo-packages.s3-website-us-east-1.amazonaws.com/3.10.17-rt12/all/repodata/repomd.xml: [Errno -2] Name or service not known
                                                                               
Fetching information for 'imx6solo_3dr_1080p'...
-> http://solo-packages.s3-website-us-east-1.amazonaws.com/.../repomd.xml      
repomd.xml                                                               [ 56%]
error: Failed acquiring release file for 'imx6solo_3dr_1080p':
error: http://solo-packages.s3-website-us-east-1.amazonaws.com/3.10.17-rt12/imx6solo_3dr_1080p/repodata/repomd.xml: [Errno -2] Name or service not known
                                                                               
Fetching information for 'cortexa9hf_vfp_neon'...
-> http://solo-packages.s3-website-us-east-1.amazonaws.com/.../repomd.xml      
repomd.xml                                                               [ 81%]
error: Failed acquiring release file for 'cortexa9hf_vfp_neon':
error: http://solo-packages.s3-website-us-east-1.amazonaws.com/3.10.17-rt12/cortexa9hf_vfp_neon/repodata/repomd.xml: [Errno -2] Name or service not known

robinabraham, 2015-10-30:
The errors I reported when running "solo install-smart" were being hit because the drone was not able to resolve "http://solo-packages.s3-website-us-east-1.amazonaws.com". Works when I changed the nameserver entry in /etc/resolv.conf to point to namserver on our corporate network.

Thanks

sjelavic, 2015-12-15:
When I issue solo wifi command in cygwin bash shell the command just hangs there and there is no response back at all. This also happens with teh solo info command. Any help appreciated.

hamishwillee, 2015-12-16:
@sjelavic Why are you using the cygwin bash shell? You should be able to install and run solo CLI in an ordinary windows python command prompt. Things to check are that you are actually connected to Solo wifi network when you do this.

sjelavic, 2015-12-16:
Thanks @hamishwillee 
Having issues when trying to install CLI in regular dos shell.
Gives me the following errors:
Exception:
Traceback (most recent call last):
  File "c:\python27\lib\site-packages\pip\basecommand.py", line 211, in main
    status = self.run(options, args)
  File "c:\python27\lib\site-packages\pip\commands\install.py", line 311, in run
    root=options.root_path,
  File "c:\python27\lib\site-packages\pip\req\req_set.py", line 646, in install
    **kwargs
  File "c:\python27\lib\site-packages\pip\req\req_install.py", line 803, in install
    self.move_wheel_files(self.source_dir, root=root)
  File "c:\python27\lib\site-packages\pip\req\req_install.py", line 998, in move_wheel_files
    isolated=self.isolated,
  File "c:\python27\lib\site-packages\pip\wheel.py", line 339, in move_wheel_files
    clobber(source, lib_dir, True)
  File "c:\python27\lib\site-packages\pip\wheel.py", line 317, in clobber
    shutil.copyfile(srcfile, destfile)
  File "c:\python27\lib\shutil.py", line 83, in copyfile
    with open(dst, 'wb') as fdst:
IOError: [Errno 13] Permission denied: 'c:\\python27\\Lib\\site-packages\\Crypto\\Cipher\\_AES.pyd'

hamishwillee, 2015-12-16:
@sjelavic I have no idea why it does that, but I had similar problems recently when I upgraded. I fixed them by simply renaming/deleting the problem _xxxx.pyd files. Nothing else has fallen over (yet) so I am assuming these are not needed.

sjelavic, 2015-12-17:
@hamishwillee  Actually the solo commands worked in dos shell after I installed the solo-cli through cygwin. But had another issue with provisions in dos shell. Had to do keygen in cygwin and then pointed to cygwin key directory by using dos $HOME env variable. If you don't do that the solo commands get stuck in connecting state.

hamishwillee, 2015-12-18:
@sjelavic I don't use Cygwin at all for this stuff. But I am using the WinPython shell and SSH from git for Windows. Anyway, did you try renaming the _xxx.pyd file as I suggested?

hamishwillee, 2015-12-18:
@tcr3dr Any ideas on how we can help @sjelavic debug this issue?

sjelavic, 2015-12-18:
I have not tried it yet @hamishwillee because dos shell works after I do the cli install from sygwin shell.
But will give it a try this weekend. Thanks for your help. BTW - have you tried running the web server in the Solo? I get some errors.

hamishwillee, 2015-12-20:
Hi @sjelavic - I haven't tried the web server yet, but I understand there is a known issue, and a patch that is expected to go in shortly.

Lybior, 2015-12-23:
@robinabraham Hello, could you please be more specific how you solved your problem? I have almost the same, thanks ; 
"Mac-mini-2:~ tms$ solo install-smart
NOTE: this process requires simultaneous access to
Solo and to the Internet. if you have not yet done so,
run solo wifi to connect to Solo and to a local
wifi connection simultaneously.

connecting to solo...
waiting for Internet connectivity...

setting up repositories.
rpmsys...
all...
cortexa9hf_vfp_neon...
cortexa9hf_vfp_neon_mx6...
imx6solo_3dr_1080p...

Loading cache...
Updating cache... ######################################## [100%]

Fetching information for 'all'...

-> http://solo-packages.s3-website-us-east-1.amazonaws.com/.../repomd.xml

repomd.xml [ 6%]

Fetching information for 'imx6solo_3dr_1080p'...
-> http://solo-packages.s3-website-us-east-1.amazonaws.com/.../repomd.xml

repomd.xml [ 31%]
error: Failed acquiring release file for 'all':
error: http://solo-packages.s3-website-us-east-1.amazonaws.com/3.10.17-rt12/all/repodata/repomd.xml: [Errno -2] Name or service not known

Fetching information for 'cortexa9hf_vfp_neon_mx6'...
-> http://solo-packages.s3-website-us-east-1.amazonaws.com/.../repomd.xml

repomd.xml [ 56%]
error: Failed acquiring release file for 'imx6solo_3dr_1080p':
error: http://solo-packages.s3-website-us-east-1.amazonaws.com/3.10.17-rt12/imx6solo_3dr_1080p/repodata/repomd.xml: [Errno -2] Name or service not known

Fetching information for 'cortexa9hf_vfp_neon'...
-> http://solo-packages.s3-website-us-east-1.amazonaws.com/.../repomd.xml

repomd.xml [ 81%]
error: Failed acquiring release file for 'cortexa9hf_vfp_neon_mx6':
error: http://solo-packages.s3-website-us-east-1.amazonaws.com/3.10.17-rt12/cortexa9hf_vfp_neon_mx6/repodata/repomd.xml: [Errno -2] Name or service not known
error: Failed acquiring release file for 'cortexa9hf_vfp_neon':
error: http://solo-packages.s3-website-us-east-1.amazonaws.com/3.10.17-rt12/cortexa9hf_vfp_neon/repodata/repomd.xml: [Errno -2] Name or service not known

Updating cache... ######################################## [100%]

Channels have no new packages.

smart package manager is now ready to use."

robinabraham, 2015-12-23:
Hi @Lybior, 

In my case, the series of "Name or service not known" errors were because the name server was not able to resolve the urls. I looked in /etc/resolv.conf and saw the entry 8.8.8.8 for Google DNS. Since I was behind corporate firewall, I changed it to 192.168.1.1 -- that fixed the issue and I was able to proceed with the installs.

As a test, see if you can ping http://solo-packages.s3-website-us-east-1.amazonaws.com/ from the solo. 

Thanks
Robin

Lybior, 2015-12-23:
@robinabraham  Thanks, you are right. I cant ping 54.231.97.210 from SOLO (just stucked on this "root@3dr_solo:~# ping 54.231.97.210
PING 54.231.97.210 (54.231.97.210): 56 data bytes
", but I can ping it from home terminal. 
("PING 54.231.97.210 (54.231.97.210): 56 data bytes
64 bytes from 54.231.97.210: icmp_seq=0 ttl=37 time=147.115 ms
64 bytes from 54.231.97.210: icmp_seq=1 ttl=37 time=139.874 ms
64 bytes from 54.231.97.210: icmp_seq=2 ttl=37 time=139.327 ms..." 
So there has to be problem somewhere between SOLO and my computer..I am just thinking if is SOLO correctly connected to internet if I am on desktop, when the wifi is connected to SOLO wifi and ethernet is connected normally. I think it should be OK, right?

Thanks for all help.. Lybior

sjelavic, 2015-12-26:
Thanks @hamishwillee I am a bit behind due to holidays. Will get back into this right after the holidays. @robinabraham not sure if I understood you right but looks like you do not have the internet access from your drone. The drone and the controller and the laptop have to be all interconnected and connected to the internet after you run the solo wifi command. I believe the internet access is through your controller.

mvpete13, 2016-02-02:
I installed solo cli with no issues at all.  But i am having problem of when i type "solo info", it just stuck at connecting.  I have read all of the comments above but i still cant figure out the reason why.  I had to press ctrl+c to stop the operation. 

C:\Windows\system32>solo info
connecting to Solo and the Controller...
Traceback (most recent call last):
  File "C:\Python27\Scripts\solo-script.py", line 9, in <module>
    load_entry_point('solo-cli==1.1.1', 'console_scripts', 'solo')()
  File "C:\Python27\lib\site-packages\pkg_resources\__init__.py", line 547, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "C:\Python27\lib\site-packages\pkg_resources\__init__.py", line 2719, in load_entry_point
    return ep.load()
  File "C:\Python27\lib\site-packages\pkg_resources\__init__.py", line 2379, in load
    return self.resolve()
  File "C:\Python27\lib\site-packages\pkg_resources\__init__.py", line 2385, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "C:\Python27\lib\site-packages\soloutils\__main__.py", line 39, in <module>
    soloutils.info.main(args)
  File "C:\Python27\lib\site-packages\soloutils\info.py", line 12, in main
    controller = soloutils.connect_controller(await=True)
  File "C:\Python27\lib\site-packages\soloutils\__init__.py", line 56, in connect_controller
    return _connect('10.1.1.1', await=await, silent=silent)
  File "C:\Python27\lib\site-packages\soloutils\__init__.py", line 33, in _connect
    client.connect(ip, username='root', password='TjSDBkAu', timeout=5)
  File "C:\Python27\lib\site-packages\paramiko\client.py", line 325, in connect
    t.start_client()
  File "C:\Python27\lib\site-packages\paramiko\transport.py", line 488, in start_client
    event.wait(0.1)
  File "C:\Python27\lib\threading.py", line 621, in wait
    self.__cond.wait(timeout)
  File "C:\Python27\lib\threading.py", line 359, in wait
    _sleep(delay)
KeyboardInterrupt

C:\Windows\system32>

does it have something to do with line 621 and 359 of threading.py?

hamishwillee, 2016-02-05:
@mvpete13 I assume you're actually connected to Solo network when you try this? Are you using the most recent solo cli and solo?

When I try this from my Windows 10 machine I get:

```
D:\Github\solodevguide\examples\helloworld>solo info
connecting to Solo and the Controller...
{
  "controller": {
    "ref": "3dr-controller-imx6solo_3dr_artoo-20151118202515",
    "version": "1.3.0"
  },
  "gimbal": {
    "connected": true,
    "version": "1.2.0"
  },
  "pixhawk": {
    "apm_ref": "331103b2",
    "px4firmware_ref": "60133536",
    "px4nuttx_ref": "d48fa307",
    "version": "1.2.19"
  },
  "solo": {
    "ref": "3dr-solo-imx6solo_3dr_1080p-20151118202712",
    "version": "1.3.0"
  }
}
```

mvpete13, 2016-02-05:
thank you for responding @hamishwillee, My pc is connected to Solo wifi.  i have been working on this issue for the longest.  im also using Windows 10.  The Solo and controller are all updated to the newest version.  I use WinPython-32bit-2.7.10.2 to download the Solo Cli.  Do you think maybe there is a incompatibility issue?  or maybe directory problems?  Do i need to add some new path in environment variables?

I have unintall and reinstall it several times.  Every time i run this code:

**_pip install -UI git+https://github.com/3drobotics/solo-cli_**

i get a error of "failed building wheel for pycryto".  So i upgrade wheel and it says "requirement already up-to-date".  I install "pycrypto-2.6.win32-py2.7" but it still "failed buiding wheel for pycrypto".  So i ran this zip code:

**_pip install https://github.com/3drobotics/solo-cli/archive/master.zip_**

i would get "Successfully installed solo-cli-1.1.1" but when i run the solo commands, it would just stuck at **_connecting_**.

Could you tell me which version of winpython you are using and did you have to install wheel and pycrypto?

Thank you again for helping me on this!!!!!

hamishwillee, 2016-02-05:
Hi @mvpete13 

I am using "the latest" WinPython , and I also use ActivePython 2.7.10. YOu shouldn't need to have ANYTHING on your computer except Python and PIp to run the solo cli. Except perhaps virtualenv if you are running the packaging code.

Your installation certainly isn't "clean". I'll see what my colleagues suggest.

mvpete13, 2016-02-07:
i tried with ActivePython and got exactly the same issue .  I wondering maybe its my env variables?

hamishwillee, 2016-02-08:
I *think* it might be that you need to have a C build environment, [based on posts like this](http://www.paramiko.org/installing.html#pycrypto). I haven't heard of such a dependency, but hard for me to be sure because I do have cygwin installed. Still asking internally.

sjelavic, 2016-02-08:
I had a similar issue. Just back from my vacation so my brain is a bit "fuzzy" but I think I had the authentication issue. If you try the solo provision command it may give you some clues. I will set up my system later and ask one of the guys here to try it with his laptop.

mvpete13, 2016-02-09:
thank you @sjelavic for responding and also @hamishwillee for your help.  I tried using cygwin like you @sjelavic and it just stuck at connecting, i think i have the same issue you had from your comment above.  i tried the procedure where you had to do keygen in cygwin and then direct the cygwin key directory by creating a HOME env variable.  It didnt work for me, obviously it was something i did wrong.  Could give me a little bit more details on how you went about doing that.  Greatly appreciated for all you guys help.

Averoteriz, 2016-02-21:
Hi all - struggling with the same issue as @mvpete13 . I've had the same errors spit out both from a Windows 7 machine in python 2.7.(several releases including 10), and on a fresh install of Ubuntu running python 2.7.10. I also tried a provision, but may have done something wrong. 
I can ssh/scp into solo with no issues. 

Any updates would be much appreciated!

hamishwillee, 2016-02-21:
@mvpete13 @Averoteriz Have you been able to SSH in at all (as per instructions)? I tried this on a clean MacOSX machine and was not getting connection on MAC initially. During SSH connection keys got exchanged. When I then tried ``solo info`` everything worked fine.

Please let me know if this was the cause/worked.

As to `solo wifi` there seems to be some dependency on your wifi system, which I hope will get resolved. I can connect quite happily through my shared internet connection on my phone, but not through the normal home wifi network. So suggest you try using your phone as a wifi hotspot (connected to your home wifi so you don't have to pay a phone bill!). If that works, it shows at least that the problem is this "known issue" (if this is the problem please post what your wifi system is)

Averoteriz, 2016-02-21:
@hamishwillee Thanks for the quick reply! I am able to ssh into both solo and the controller without issue, but any of the solo cli commands result in the same error. I only tried this on windows - will report back as to using Ubuntu. Nice idea with the alternate wifi - I'll give that a go when I get the cli talking.

Averoteriz, 2016-02-21:
@hamishwillee HEYYYY! Worked fine on Ubuntu after ssh'ing in first -  I was able to establish a wifi connection to my router (Arris DG1670). I'll have to dig into the problem a little more for Windows at some point. Thanks for the tips!

hamishwillee, 2016-02-22:
Great @Averoteriz ! FYI, you shouldn't need to do this connection before using solo-cli. Investigation to follow here: https://github.com/3drobotics/solodevguide/issues/288

If you find out your router is part of the problem, that discussion to be carried on here: https://github.com/3drobotics/solodevguide/issues/289

Are there any open issues still on this thread? If not, I'd like to close it.

mvpete13, 2016-02-23:
@Averoteriz @hamishwillee, sorry for the late reply.  I still couldnt get cli to work in Windows.  I am yanking all my hair out.  i guess i'll have to try using Ubuntu.  I still dont understand how i can connect to the Solo network and connect using WinSCP or Putty but couldnt connect to CLI.  I have been doing a lot of researching online and many people did not have this problem.  Just like them, I have Windows 10 and the most updated version of CLI and Solo.  I even used the same version of WinPython as them, but it still does'nt work.  I have download, delete, and redownload CLI countless times.  I just dont get it.  I gonna keep trying maybe until i go insane, but if i find anything new i will definitely post it.  @hamishwillee, please dont close the thread just yet, give me a little bit more time.  Thank you.

hamishwillee, 2016-02-23:
@mvpete13 Of course we'll keep it open! What version of solo-cli are you on now? The latest version has some changes. Can you try solo-info latest version again and dump in the output of ``solo info``. I know you did this already, but there is a lot of discussion here now.

hamishwillee, 2016-02-23:
FYI, we've updated the solo-cli again with fixes to packaging on Windows.

[4 more replies]