3drobotics/solodevguide #175: Python and Pip installation note in "installing Solo CLI" should be linked to a good place to install from

tcr3dr, 2015-09-25:
**Author:** Kelsey
**Page:** http://dev.3dr.com/starting-utils.html#connecting-to-solo-and-the-internet

Python and Pip installation note in "installing Solo CLI" should be linked to a good place to install from

tcr3dr, 2015-09-25:
Good point. @hamishwillee I feel this is something that's probably already been written before (DroneKit? MAVProxy?) Can we borrow some copy for this?

hamishwillee, 2015-09-28:
@tcr3dr Well, we cheated a bit for DK on Windows by concentrating on only one Python distribution - WinPython (for good reasons related to dependencies present in that distribution). 

1. I presume we only support Python 2?
1. If it is OK to concentrate on just a few distributions ... how about (after *You will need Python and pip installed in order to run this utility.*) adding the following **note**?

> *  On Windows we recommend [ActiveState Python](http://www.activestate.com/activepython/downloads) or [WinPython](http://winpython.github.io/) which come with PIP installed. PIP is included on all Windows distributions starting from Python 2.7.9.
> * Most Linux distributions include Python 2. On Ubuntu Linux you can additionally install PIP with the command line:
```
    sudo apt-get install python-pip 
```
> * Mac OS X 10.8 comes with Python 2.7 (including PIP) pre-installed by Apple.