3drobotics/solodevguide #176: Collecting but not using email addresses submitted through this Olark – do we need these email addresses? Can we get rid of the field? (is the name field even useful? maybe it would be better to ask for a GH username there)

tcr3dr, 2015-09-25:
**Author:** Kelsey
**Page:** http://dev.3dr.com/starting-utils.html#connecting-to-solo-and-the-internet

Collecting but not using email addresses submitted through this Olark – do we need these email addresses? Can we get rid of the field? (is the name field even useful? maybe it would be better to ask for a GH username there)

tcr3dr, 2015-09-25:
@Frijol, the email addresses are collected by the owners of the Olark message and I do use them for direct responses if necessary.

A Github username would be useful, perhaps it's either Github *or* email?

hamishwillee, 2015-09-28:
What would also be nice is a form that returns the Github issue number that was created. If you're accepting github ids (a good idea) then using that ID as part of the text so their interest is registered would be good.

tcr3dr, 2015-10-28:
Closing in favor of #101. Feedback should be added there.

closed 2015-10-28 by tcr3dr