3drobotics/solodevguide #179: What happens when you give the wrong password to solo wifi?

tcr3dr, 2015-09-25:
It should be much more descriptive.

franknitty69, 2015-10-03:
Hello,

i am definitely giving the correct password when i execute solo wifi, but i get Sending discover repeatedly. it keeps going until i reboot the controller. my sid has a space in it if that matters i.e. My SID

My-Computer:~ my-username$ solo wifi --name=“My SID" —password=mypassword
connecting to the Controller...
restarting DNS forwarder and DHCP server: dnsmasq... 
stopping DNS forwarder and DHCP server: dnsmasq... stopped /usr/bin/dnsmasq (pid 792)
done.
starting DNS forwarder and DHCP server: dnsmasq... done.
done.
Successfully initialized wpa_supplicant
rfkill: Cannot open RFKILL control device
udhcpc (v1.21.1) started
Sending discover...
Sending discover...
Sending discover…

hamishwillee, 2015-10-04:
@franknitty69 Possible duplicate of #156

tcr3dr, 2015-10-28:
@franknitty69 The expected behavior now is you will see a log of it having tried for 10 seconds, and then if it fails, it will give up and notify you of such.

Will close this in favor of #156.

closed 2015-10-28 by tcr3dr