3drobotics/solodevguide #186: Make it clear which version of Python Solo runs

tcr3dr, 2015-09-28:
`2.7.3`

hamishwillee, 2015-09-29:
Where would you put this information? Generally we're advising people to install their own environment rather than use Solo's, and  there isn't anywhere yet for "complete specification of the standard software stack" that this might otherwise fit. 
Or is there somewhere it is specifically needed?

tcr3dr, 2015-10-28:
@hamishwillee This seems like, perhaps, a System Diagram issue. Like, we are running a ARM core (which arm core?), Yocto Linux, using this version of Python, this many cores, etc. http://dev.3dr.com/concept-architecture.html Seems appropriate.

hamishwillee, 2015-10-29:
@tcr3dr That works for me. 

I can see that we're using "3DR Poky (based on Yocto Project Reference Distro) 1.5.1 \n \l" (is this 3DR Poky 1.5.1 OR based on  Yocto Project Reference Distro 1.5.1?)

Can you point me to the other information we'd add in this section - cores, other key software stack versioning? Is there an easy way for people to generate a log of this stack so that we don't have to go to lots of effort keeping it up to date?

tcr3dr, 2015-10-31:
@hamishwillee The stack will likely not change, so no worry about publishing this at first:

1. 3DR Poky (based on Yocto Project Reference Distro, Poky 1.5.1)
2. ARM Cortex A9 (i.MX6 Solo by Freescale), 1Ghz, 1 CPU core with VPU and GPU
3. Python 2.7.3

I like the idea of generating a stack from the command line, might be worth looking into. At the very lease, `uname -a` gives you 1) and `python --version` gives you 3)

hamishwillee, 2015-11-06:
@tcr3dr Well, I created #216 to provide what information we have now. This is in a new section of the page "Solo hardware/software stack" - OK? It feels a bit odd because there are only three items, and presumably there is a lot more to the stack than that.

Anyway, it provides the information that kicked this discussion off, and once the title is in there, we can extend the information as you think of new things to include.

I haven't put in a section on how you can get this information/list it.