3drobotics/solodevguide #190: Developper Guide Needs a Dedicated Section for Controller

emogenet, 2015-09-30:
Currently, when you read the guide, it is very focused on the solo itself whereas the system
is essentially comprised of two fairly different devices.

First, it is at times unclear whether the instructions apply to the controller or to the solo itself.

Second, there should be a dedicated section for the controller. For example, it is currently
almost impossible to update the firmware on the controller without the solo connected, and
the guide doesn't really help with figuring it out.

hamishwillee, 2015-09-30:
@emogenet Thanks for your post. None of the instructions apply to the controller, other than to explain how to connect to it. On the [first topic](http://dev.3dr.com/starting-network.html#using-ssh) there is a note:

> We do not recommend modifying the Controller. All examples in this guide will be focused exclusively on modifying Solo.

@tcr3dr 

1. Perhaps we should add this note to the front page of the docs?
1. We probably do need a topic "Updating Solo firmware" and "Updating Controller Firmware"
1. The above implies that we probably also need an icon to differentiate Controller commands from Solo or PC commands

tcr3dr, 2015-09-30:
@hamishwillee Actually, there are no commands which are intended to be run on the Controller. If there are, I'd consider this a bug or unclear wording.

@emogenet Thanks for cross-posting here. I agree that we need a dedicated controller section, especially to make the above points more clear.

Specifically regarding upgrading firmware: we have an open issue https://github.com/3drobotics/solo-cli/issues/8 for enabling firmware updating from the Solo CLI. This will be as simple as `solo update solo latest`, `solo update controller latest`, or `solo update both latest`. We should have this published in a matter of days (sorry for the delay). It's my preferred method of updating the firmware (vs. the app) and I think you'll feel likewise.

As for documenting the Controller, are you interested in system level diagrams and technical details? Both of those can and should be added. To do more than that, i.e. fully support running custom code on the Controller, requires setting up another package repository dedicated to the Controller. This might be really useful! But in the short term, this is lower priority than improving the Solo-focused dev experience. The more specific your goals (running custom Python code, running DroneKit code, etc.) the more easily we can start documenting those steps into a Controller-focused section.

tcr3dr, 2015-10-03:
@emogenet The newest version of solo-cli adds `update` and `revert` for the controller and for solo. Reinstall from https://github.com/3drobotics/solo-cli and test it out.

hamishwillee, 2015-10-04:
@tcr3dr That's cool. I see that this update/revert is documented in the [solo tool page already](http://dev.3dr.com/starting-utils.html). Some questions:

1. Is it worth making this more prominent? My thought is that:

   *  it is worth having a controller page that covers update/revert and the fact that we don't cover much else about congtroller. 
   * Is updating firmware a reasonable first step to add to [troubleshooting](http://dev.3dr.com/starting-troubleshooting.html)

2. With respect to our current docs my thoughts were:

   * How can I list the possible versions available?
   * If there is a non-developer way to update the firmware?
   * Are there preconditions - presumably usual "be connected to solo and solo to home wifi?
   * On "revert" what is difference between current and latest?
   * Whether or not we have a dedicated page that we link to for update/revert, it would be good to have  a few words about the update option in the solo tools page.

ShawnSchaerer, 2016-01-20:
@tcr3dr any progress on the controller documentation?

hamishwillee, 2016-01-20:
I can answer that. Not yet.

wenzelsolo, 2016-01-31:
I have a 3dr solo that has a level error I have tried fixing on the solo app and mission planner but no success. I have remove the controller. how do I go about fixing it?

wenzelsolo, 2016-01-31:
the controller p/n is ph11a5603850.

franknitty69, 2016-01-31:
Wenzelsolo, this site is for development and this issues section is for issues regarding development.

If you are having issues with your controller please contact support. The contact info is at https://3dr.com/support.

Also see please see the kb and user manual at https://3dr.com/kb/solo. Calibration and pairing is covered there.

proficnc, 2016-01-31:
By the look of that part number, that's the Pixhawk 2.

Please call customer support for assistance.

wenzelsolo, 2016-01-31:
I look into the 3dr solo calibration but it didn't help. There is instructions to update pixhawk 2 via usb plug. But there is instruction  for pixhawk. Goin forward do what is the factory firmware for pixhawk 2? Apm copter v3.2.3?

proficnc, 2016-01-31:
Call service!  This is not the place!

wenzelsolo, 2016-01-31:
I did they haven't call me back.

Thanks anyway

hamishwillee, 2016-01-31:
@wenzelsolo Try them again. This github repo is for issues with the Solo Development Guide, and specifically this particular issue is about controller developer information.

wenzelsolo, 2016-01-31:
I understand that thank you

wen