3drobotics/solodevguide #192: License clarifiation

LorenzMeier, 2015-10-06:
As people start to contribute, I wanted to make sure that the license was intended to be CC-BY-SA - @liamstask is that the case? It would have ramifications for 3DR's ability to use content from here in other documentation bits, as all these would become CC-BY-SA, too.

liamstask, 2015-10-06:
good call. @tcr3dr - let's figure out which license makes sense, and add it both to the repo and to [CONTRIBUTING.md](https://github.com/3drobotics/solodevguide/blob/master/CONTRIBUTING.md)

hamishwillee, 2015-10-06:
Also consider that some bits might need to move into dronekit docs at some point - so even if you decide that the licence can't be CC-BY-SA it should be compatible with CC-BY-SA.

tcr3dr, 2015-10-29:
We have decided to relicense as CC-BY 4.0 to make our documentation more versatile for downstream groups. This issue is open to track that progress.

tcr3dr, 2015-10-29:
Completed. https://github.com/3drobotics/solodevguide/pull/209

closed 2015-10-29 by tcr3dr