3drobotics/solodevguide #196: hi there.  i'm trying to write my own android app to control solo.  is it possible to access video stream directly from my own android app?  thanks.

tcr3dr, 2015-10-09:
**Author:** duy huynh
**Page:** http://dev.3dr.com/

hi there.  i'm trying to write my own android app to control solo.  is it possible to access video stream directly from my own android app?  thanks.

djnugent, 2015-10-12:
If you are using dronekit android then you should have access to the video. Tower has a working example using 3DRservices and dronekit android. I am not familiar with the code base but you can poke around here: https://github.com/DroidPlanner/Tower/search?utf8=%E2%9C%93&q=video

djnugent, 2015-10-13:
Do responses notify the original author?

tcr3dr, 2015-10-28:
@djnugent They don't. I'll respond to them directly. This is also generally a forum question, rather than a deficiency with the docs.

closed 2015-10-28 by tcr3dr