3drobotics/solodevguide #198: Sidebar corruption

djnugent, 2015-10-12:
The sidebar loses items after an item is clicked. The only way to restore it is to refresh the page.

[image]

tcr3dr, 2015-10-12:
@djnugent What are the full steps to reproduce this?

djnugent, 2015-10-12:
Navigate to main page. Click on an item in the sidebar.

I can only get it happen on Win 10 chrome. Let me try to restart chrome.

djnugent, 2015-10-13:
Issue persists with restart. Chrome Version 45.0.2454.101 m

hamishwillee, 2015-10-13:
@djnugent Based on your screenshot this is (IMO) a variant of #121 - essentially if you have a search term it persists even through refreshes and that is why only some of the items are shown (see screenshot - you have "git" there).

The "workaround" is to delete the search term. This is broken though!

Close if you agree!

djnugent, 2015-10-13:
Yes that is the issue.

hamishwillee, 2015-10-13:
Closing. Dupicate of #121

closed 2015-10-13 by hamishwillee