3drobotics/solodevguide #199: solo wifi command expecting Controller version to be be 1.1.15

franknitty69, 2015-10-17:
solo can't connect to the internet with v1.2

**steps to reproduce**
1. connect computer to solo wifi
2. open terminal and execute solo wifi --name=SID --password=password

**result**
*error: expecting Controller to be EXACTLY EQUAL TO 1.1.15
your Controller version: 1.2.0
please flash your Controller with 1.1.15 to run this command
solo update solo 1.1.15
solo update controller 1.1.15*

**expected result**
solo should connect to the wifi indicated in the solo wifi command

tcr3dr, 2015-10-28:
@franknitty69 Thanks for the good bug writeup!

solo-cli has been updated to support 1.2.0 and above. The original implementation of the command didn't deal with differing wifi channels, so we had to rework it. The new version now conencts to your local network, then re-establishes connection with Solo. It should address a lot of the earlier issues.

Install from [the solo-cli repo](http://github.com/3drobotics/solo-cli) or just run:

```
sudo pip install -UI git+https://github.com/3drobotics/solo-cli
```

Let me know if this works for you!

tcr3dr, 2015-10-28:
Closing since we now support 1.2.0 and above. #156 is open for issues relating to using `solo wifi`.

closed 2015-10-28 by tcr3dr