3drobotics/solodevguide #202: Could you please add to this page to make it more friendly to new-ish users? Right now in the app you can email logs to yourself, but this functionality is going away in the next version of the app (they go directly to zendesk) so I'd like to be able to direct people to this page if they want to pull their own logs.

I think it just needs these instructions, formatted better:

mkdir solo controller
scp root@10.1.1.1:/log/* ./controller
TjSDBkAu
scp root@10.1.1.10:/log/ ./solo
TjSDBkAu

tcr3dr, 2015-10-26:
**Author:** Nick
**Page:** http://dev.3dr.com/starting-troubleshooting.html

Could you please add to this page to make it more friendly to new-ish users? Right now in the app you can email logs to yourself, but this functionality is going away in the next version of the app (they go directly to zendesk) so I'd like to be able to direct people to this page if they want to pull their own logs.

I think it just needs these instructions, formatted better:

mkdir solo controller
scp root@10.1.1.1:/log/* ./controller
TjSDBkAu
scp root@10.1.1.10:/log/ ./solo
TjSDBkAu

hamishwillee, 2015-10-26:
This sounds like a good idea - at the moment all we do is say "this is where the logs are" but not how you get them worth spelling out. 

BUT SCP is a unix/linux command - what would be the best recommendation for Windows users?

eliao, 2015-10-26:
putty/pscp?

hamishwillee, 2015-10-26:
Probably easier than rsync. Nothing built in?

franknitty69, 2015-10-27:
Windows requires an ssh or sftp/scp client. Putty gives you command line interface. Winscp gives you a UI where you can connect, browse and drag logs from artoo to your desktop.

hamishwillee, 2015-10-27:
Probably putty - we already [recommend it here](http://dev.3dr.com/starting-network.html#using-ssh). Would the instructions be exactly the same as [above](https://github.com/3drobotics/solodevguide/issues/202#issue-113429133) except replacing scp with pscp?

I can't test this  - don't have solo yet. If I put together a new version of the docs can someone volunteer to test them?

franknitty69, 2015-10-27:
the instructions are the same for scp or pscp.

i can test as well.

tcr3dr, 2015-10-28:
I mentioned a while ago, but I would love to simply expose this via solo-cli using `solo logs download`: https://github.com/3drobotics/solo-cli/blob/master/soloutils/logs.py

This is separate from documenting *where* the files are, which we need to do, but would hopefully be more straightforward for someone doing maintenance to want to dump their log files.

Right now `solo logs` is very basic, but can be expanded to include the Controller, or more files if necessary.

hamishwillee, 2015-10-29:
@tcr3dr Yes, sounds like a very good idea from a KISS point of view. So 

1. does this ``solo logs`` all work right now (ie is this in the version of the cli that people get when they install the utils
1. As I understand it, the logs will all just get copied to the PC running the command, in the subdirectory /logs/ below the current directory? Correct?

tcr3dr, 2015-10-29:
It works right now, downloading the contents of Solo's `/log` into the current folder.

I would like it to do `./drone` and `./controller`, probably. But that's a simple enough change.

hamishwillee, 2015-10-29:
So all logs, from controller and drone currently go into the same /logs folder?

tcr3dr, 2015-10-30:
Right now, `solo logs download` rsyncs all files on Solo's `/log` folder in your local directory `.`.

tcr3dr, 2015-10-30:
Nevermind, latest version downloads into `./drone` and `./controller` all your log files.

nick3dr, 2015-11-25:
I'd like very simple and self-contained instructions for tech support to be able to point people to, in response to the question "How do I see my own logs"

I'm pretty unfamiliar with the "solo logs" command and where one types that. 

The nice thing about scp (at least for unix users) is that it's just 1 command you need to type. No setup needed

hamishwillee, 2015-11-25:
@nick3dr I addressed this 27 days ago as #210. See [here](https://github.com/3drobotics/solodevguide/blob/hgw_download_logs_202/book/starting-troubleshooting.md) and [here]().

This has a stand-alone topic as part of solo tool command for how to download logs. @tcr3dr can we get that merged - whether or not it is the final version it is much better than what we have now.

I see your points above and I strongly disagree. My argument goes like this

1. You already have to install solo tools to develop with this platform. I'm not sure you can just use SCP without having first set up solo tools ... though I could be wrong
1. Its just one command that does everything ... on all platforms. The instructions explain very clearly what you need to do.
1. We're not just targeting an audience of unix users. Until this year I'd never heard of the tool. 

All that is not to say this is perfect. We certainly could consider pulling this topic out of troubleshooting. At that point having sections with different options/ways of getting the logs might be reasonable. What do you think @tcr3dr ?

franknitty69, 2015-11-26:
You can use WinSCP w/out using solo cli. It's an explorer style SCP Tool. Just connect your computer to solo wifi, in WinSCP browse to /log and copy the files to the laptop.

Unfortunately there isn't a simple method for retrieving logs that support can refer users to. Its either use solo cli via terminal/winpython or use sftp/scp/rsync via terminal/winscp.

hamishwillee, 2015-11-26:
@franknitty69 Thanks. WinSCP sounds pretty handy, though I probably wouldn't cover it in great detail since there is no "legacy loyalty" to that approach in the Windows community. So it will be Solo logs as first option, and further option of SCP ... if Nick chooses to provide basic instructions. 

I've just got hold of a Solo, so I can at least now verify stuff!

nick3dr, 2015-11-26:
My point is that there is a large community of people who want to review
flight logs but do not want to develop on Solo. We recently took the
functionality of emailing yourself logs out of the app (it was mostly an
accidental feature that went away as the support ticket submission process
has improved), and it would be very good to provide a minimum-friction
method for these users to get their logs with bare minimum setup. I agree
that unix-only is insufficient, but I don't think that users need to have
heard of the tool before as long as there is a simple command they can copy
and paste into the terminal to get their logs.

hamishwillee, 2015-12-02:
Hi @nick3dr 

I have updated this section in #210. YOu can get an idea of what it will render like [from here](https://github.com/3drobotics/solodevguide/blob/hgw_download_logs_202/book/advanced-logs.md).

In essence I have split the end-user stuff into its own section and made it clear that different approaches suit developers vs end users. The information about what the logs "are" is at the end, because end users won't be all that interested. The information is linked from the beginning of the article though, so developers can find it easily. 

Can you sanity check it?

PR #210 (2015-12-14): Fixes #202
This fixes #202 - essentially it documents how to use the feature in solo cli document and links to this from the Troubleshooting topic.

@tcr3dr Most of the detailed information about logging is actually not all that appropriate/useful in "Troubleshooting".  
I propose that we move this content into its own page in "Advanced Topics" under page title "System Logs". We still need information about logs in Troubleshooting but this can just be  the existing heading and the new link to the solo cli. Instead of all the detail we have one line saying something like "System logs can help d [... cut]
commit 5d9c27ac (2015-12-14): Merge pull request #210 from 3drobotics/hgw_download_logs_202

Fixes #202
commit 82248afa (2015-12-14): Fixes #202
closed 2015-12-14 by mrpollo