3drobotics/solodevguide #208: /dev/video0 in Video Pipeline

vitorguizilini, 2015-10-29:
Hi,

I am trying to access the GoPro live stream from my computer, but so far I haven't been able to, either using VLC or the "solo video acquire method". Everything seems to work fine, I can see the stream from my smartphone, but once I do:

$ solo video acquire
connecting to solo...
acquiring video feed...
/dev/video0 now available for use by user scripts.

The smartphone stream freezes (which I think should happen) and nothing shows up on my computer. Creating the sololink.sdp file also didn't work.

Any suggestions would be greatly appreciated.

brandongalbraith, 2015-11-11:
Seeing the same issue.

robinabraham, 2015-12-10:
Running "solo video acquire" does not freeze the smart phone stream for me. That stream stays alive. As a result, when I run any code that accesses /dev/video0 on Solo, I see the following:
`HIGHGUI ERROR: libv4l unable to ioctl S_FMT`
`HIGHGUI ERROR: libv4l unable to ioctl VIDIOCSPICT`
which indicates that /dev/video0 is busy.

mpetkova, 2016-02-17:
I can't access the stream either, but I'm getting the error:

`Warning, camera failed to properly initialize!`

I can see the video through the solo app however.

FlyingCatAlex, 2016-03-20:
I have the same issue with robinabraham. Any one figured out the reason?

franknitty69, 2016-03-21:
gents, there is an issue with the video solution that was previously listed on the dev guide. 3dr is aware of this and is working on a video proper api based video solution.

see referenced issues:

[213](https://github.com/3drobotics/solodevguide/issues/256)
[256](https://github.com/3drobotics/solodevguide/issues/256)

@hamishwillee can this be categorized as a duplicate?

mpetkova, 2016-03-21:
It was suggested to me to use libsolocam.so from the opencv examples in the meantime. Although I still can't view the live video stream, I can access it frame by frame and analyze it by making some simple changes to SoloCamera.py.

jimmy1409, 2017-09-13:
I have tried sololink.sdp to capture an image.

Sometimes, I got clear image and sometimes I am getting blurred image. I guess if I can get access to /dev/video0 , then I will be able to get clear image. I want to do object detection.

Is there any way I can access /dev/video0 in python script?

Thanks,

Dhruvil