3drobotics/solodevguide #213: Error creating multiple video processing endpoints

franknitty69, 2015-10-31:
Following the steps in the topic Video Pipeline Internals (Advanced), issuing the modprobe v4l2loopback exclusive_caps=0,0 video_nr=1,2 results in a FATAL error. i've even installed gst-plugins-good-video4linux2 via smart. same error.

**steps to reproduce**
1. connect to solo wifi
2. on your computer issue the *solo video acquire* command
3. ssh to solo and issue the *modprobe v4l2loopback exclusive_caps=0,0 video_nr=1,2* command

**result**
modprobe: FATAL: Module v4l2loopback not found.

**expected result**
v4l2loopback driver should be activated

franknitty69, 2015-11-03:
resolved by manually installing v4l2loopback (smart install v4l2loopback)

Lybior, 2015-12-24:
@franknitty69 Have you find solution for this? Thanks..

franknitty69, 2015-12-24:
No, the problem is still occurring. I even talked to developer of video4linux2. Issue is on the solo side.

GPSFan, 2016-01-14:
Chances are this is because the kernel version running on the solo is not the same as the kernel version that v4l2loopback.ko was compiled for.
smart wants to install:
  kernel-3.10.17-rt12-1.0.0-ga+g3f15a11-3.10.17-r0@imx6solo_3dr_1080p                                      
  kernel-image-3.10.17-rt12-1.0.0-ga+g3f15a11-3.10.17-r0@imx6solo_3dr_1080p                                
  kernel-module-v4l2loopback-0.9.1-r0@imx6solo_3dr_1080p                                                   
  v4l2loopback-0.9.1-r0@imx6solo_3dr_1080p     

but the kernel on my solo is:
root@3dr_solo:~# uname -a
Linux 3dr_solo 3.10.17-rt12-1.0.0_ga+g4176d57 #8 SMP PREEMPT Wed Nov 18 20:27:35 UTC 2015 armv7l GNU/Linux

Lybior, 2016-01-27:
Is it hard to compile it right? Anyone capable of doing so?

GPSFan, 2016-01-27:
It is easy to compile once you have the specific kernel version .config, patches and source. Most of which are available. I compiled the older kernel from the source provided by 3dr, as well as v4l2loopback.ko. Unfortunately as stated before, that is not what is running on Solo so unless I want to replace the kernel and ALL the modules or revert back to what Solo was running before the 1.3.0 update I/you/anyone needs EXACTLY what 3dr used as source.
I opened issue 256 on solodevguide GitHub, but have yet to receive any response from 3dr.