3drobotics/solodevguide #220: Can't view live video with VLC

squilter, 2015-11-22:
The guide says to make a file with these contents and to open it with VLC:
```
c=IN IP4 10.1.1.1
m=video 5600 RTP/AVP 96 
a=rtpmap:96 H264/90000
t=0 0
```

This did not work when I tested it on Arch LInux and then on Windows 10.  Not sure why.  It's hard to get useful debug messages out of VLC.

Does anybody know if this works with Ubuntu?
Is there some sort of multicast configuration that needs to be done first?

djnugent, 2015-12-09:
What did you do to fix this?

squilter, 2015-12-09:
I didn't fix it properly, I just used a crappy hack. See the readme at http://github.com/squilter/target-land

squilter, 2015-12-09:
Did you also experience this same problem?  If so, what OS?

djnugent, 2015-12-09:
I had it working with an Ubunutu VM. I didn't see you had to hack a solution together. I will give it a try again.

squilter, 2015-12-09:
I suspect that it only works with Ubuntu, because multicast is configured properly by default on Ubuntu systems.

mpetkova, 2016-02-17:
I'm having this issue on OS X 10.8.5.

hamishwillee, 2016-02-18:
@tcr3dr @mrpollo @djnugent Any thoughts on this one? It sounds like this is dependent on multicast being set up properly, and our instructions only cover this for Ubuntu.

Can we get more generic instructions or should we just change the title of http://dev.3dr.com/concept-video.html#viewing-live-video to "Viewing live video on Ubuntu"?