3drobotics/solodevguide #224: Log files in solo vs controller

hamishwillee, 2015-11-30:
Solo/controller both have lots of logs. There seems to be quite a lot of similarity between some of the files and in other cases there appears to be a similar file type/extension but different filename. Check out the pdf file comparison - [report1.pdf](https://github.com/3drobotics/solodevguide/files/46875/report1.pdf)

- Does the controller copy and rename some files/is there an overlap
- Do you need both controller and solo logs. @nick3dr your PDFs only explain how to get the controller logs, does that mean you think only the controller logs are useful?
- If a controller and solo both have a log of the same type, what is the difference?

I've created a single example of each log in both controller/drone below. 

- What are each of these types
- Are any completely useless to anyone - ie temps that should be ignored
- Are any duplicates on controller/drone
- Are any useful only to 3dr

#### controller
3dr-pairing.conf
3dr-solo.log.1
3dr-stm32.log.1
3dr-temp.log.1
3dr-top.log.1
3dr-video.log.1
3dr-wifi.log.1
boot.1
dmesg.1
messages.1
solo.tlog.1
wtmp.1

#### drone

3dr-rc.log.1
3dr-solo.log.1
3dr-telem.log.1
3dr-temp.log.1
3dr-top.log.1
3dr-wifi.log.1
boot.1
dmesg.1
messages.1
shotlog.000.log
shotlog.log.1
wtmp.2

eliao, 2015-11-30:
shotlog.log.n are shot manager logs.  shotlog.n.log are symlinks to these files for backwards compatibility, so we don't need those.  

3dr-telem.log.n contains some logging on bidirectional telem bandwidth.

3dr-top.log.n contains cpu usage statistics.  

On the controller:
solo.tlog.n are the telem logs that the controller has received.  

@johnfinley may be able to add some more details.

hamishwillee, 2015-11-30:
Thanks @eliao -  I copied your information above to make it clear which ones we still don't know about:
- when you say we don't need the symlinks, should we arrange these be deleted?
- Are there any documents around on interpreting the logs you have identified above (other than the tlogs of course)
- When you say "solo.tlog.n are the telem logs that the controller has received." - are you suggesting that all tlogs get copied from the Solo, and are these the SAME as 3dr-solo.log.n ? (and is the mapping 1:1 for the number - I presume not, since the controller might not have the latest, so would be one number lower)

eliao, 2015-12-01:
- symlinks can be deleted from any archives of the logs.  But we should keep them on vehicle for backwards compatibility.  
- I don't know of any docs on interpreting these logs.  They were mainly developer facing and as such used to log data for debugging.  
- I don't think we store tlogs on Solo.  Only on the controller.  The difference would be that the controller logs would only log telem that it received, so it would skip drops.

tarazansafak, 2017-03-16:
Hey guys,

Is real time IMU logs available on Solo? Can I display them via command line? Or any hints? I need to get them programatically.