3drobotics/solodevguide #225: issue while running example.py

grayjean, 2015-12-01:
```shell
(env)root@3dr_solo:~/MyPythonCode# python example.py
Traceback (most recent call last):
  File "example.py", line 1, in <module>
    from droneapi import connect
ImportError: cannot import name connect
```

I used "pip install droneapi"  still the error is occuring

hamishwillee, 2015-12-01:
@grayjean Thanks for pointing that out. The docs  in [Connecting to Solo](http://dev.3dr.com/example-dronekit.html) clearly need to be updated.

Try using "dronekit"

```
from dronekit import connect
```

You should also update from dronekit==2.0.0rc1 to dronekit==2.0.0

@tcr3dr This one a fix for you or me?

grayjean, 2015-12-01:
Thank you for the help I also have few more doubts

1) Does ```pymavlink-1.1.64``` work with this?
2) Is the following line correct?
```python 
vehicle = connect('udpin:0.0.0.0:14550', await_params=True)
``` 
I was getting an error about ```await_params``` so i changed it to ```wait_ready```
I am still getting an error
``` shell
(env)root@3dr_solo:~/MyPythonCode# python example.py 
ERROR LOADING MAVNATIVE - falling back to python implementation
>>> Link timeout, no heartbeat in last 5 seconds
>>> No heartbeat in 30 seconds, aborting.
Traceback (most recent call last):
  File "example.py", line 2, in <module>
    vehicle = connect('udpin:0.0.0.0:14550', wait_ready=True)
  File "/home/root/MyPythonCode/env/lib/python2.7/site-packages/dronekit/__init__.py", line 2354, in connect
    vehicle.initialize(rate=rate, heartbeat_timeout=heartbeat_timeout)
  File "/home/root/MyPythonCode/env/lib/python2.7/site-packages/dronekit/__init__.py", line 1815, in initialize
    raise APIException('Timeout in initializing connection.')
dronekit.APIException: Timeout in initializing connection.
```
I want to know whether  ```udpin:0.0.0.0:14550```  is correct parameter since you mentioned that docs in [Connecting to Solo] (http://dev.3dr.com/example-dronekit.html) have to be updated

hamishwillee, 2015-12-01:
> 1) Does pymavlink-1.1.64 work with this?

Probably. I haven't tested on Solo but on windows I'm using 1.1.62

> 2) Is the following line correct? vehicle = connect('udpin:0.0.0.0:14550', await_params=True)

As you say, this needs to use ``wait_ready``. I believe the following is correct:

```
vehicle = connect('udpin:0.0.0.0:14550', wait_ready=True)
```

The error says what it means - you're not getting a heartbeat, so it is failing. The string is correct AFAIK. I'm hoping @tcr3dr will have some advice for you ... if not I plan to test and update this section in the next few days.

hamishwillee, 2015-12-02:
Hi @grayjean 

I have updated the example, but I still can't get this to work. You can check out the changes on my branch for [example.py](https://github.com/3drobotics/solodevguide/blob/hgw_fix_dk_example/examples/dkexample/example.py) and [requirements.txt](https://github.com/3drobotics/solodevguide/blob/hgw_fix_dk_example/examples/dkexample/requirements.txt)

I'll be following up internally and hope to get back to you with more information.

hamishwillee, 2015-12-03:
Hi @grayjean 

This is an error in Solo 1.3.0. You can either downgrade a version (using the solo CLI tool) - any other version will work, wait until 1.3.1 comes out (probably next week), or patch the current version.

I did this in WinSCP, but basically you need to edit **<root>/usr/bin/runMavproxy.sh** and replace the contents with the following text (the last line and the trailing stash on the preceding line are what have been added)

```
#!/bin/sh

mavproxy.py \
--master=udpout:127.0.0.1:14560 \
--nolog \
--native \
--streamrate=-1 \
--daemon \
--out=127.0.0.1:14550
```

I tested this and it works. There was an exception at the end of the script, which you can ignore (this should just be absorbed - we're looking into it): ``Exception in thread Thread-1 (most likely raised during interpreter shutdown):``

Please confirm this works for you!

grayjean, 2015-12-04:
Yes it is working for me as well.
Thank you again for all your help. I very much appreciate it

hamishwillee, 2015-12-04:
You're most welcome - we appreciate the feedback/information so we can get this stuff fixed!

closed 2015-12-04 by hamishwillee