3drobotics/solodevguide #226: Smart Shots Example

franknitty69, 2015-12-01:
any word on when a legit example for smart shots will be fleshed out?

i'm trying to work one out on my own using the existing smart shots (infiniCable, cable cam, orbit) as an example. but i have some questions.

in the example, solo.smartshot is passed to the class, in solo smart shots, the class doesn't take any parameters? which is correct?

in the example, self, vehicle and shotmanager are initialized, in the solo smarts shots only self and vehicle are. which is correct?

and finally, aside from using ssh or coding a custom gcs, how can you invoke your smart shot from the solo app?

p.s. if someone has an example a dronekit smart shot, they would be extremely helpful.

hamishwillee, 2015-12-01:
> any word on when a legit example for smart shots will be fleshed out?

No. This is probably a job for me, but its in a queue. These are all good questions that I will answer when I get to it.

Hopefully someone else viewing this can respond in the meantime.

franknitty69, 2015-12-03:
here's my go at it. i tested it in sitl by copying the handle_rcs portion and related definitions into a standard dronekit python script. it works there.

if you don't mind when you get a chance can you look at it and see if anything jumps out at you. i'm going to try and test on solo when i get some good weather.

[autopan.pdf](https://github.com/3drobotics/solodevguide/files/50360/autopan.pdf)

hamishwillee, 2015-12-03:
Hi @franknitty69 

I haven't gone far enough in my investigations to offer any sort of review. I'll let you know when I do though.

Cheers
H

hamishwillee, 2016-01-05:
Hi @franknitty69 

I'm keeping this open. Yes, there has been no word on this, but I still want to create an example.

franknitty69, 2016-01-05:
ah sorry about that. i was going through and cleaning up open issues. i figured this was more of a question than an issue, but i can see why keeping it open makes sense. can't wait to see what you come up.

irishguy, 2016-01-05:
Not sure if there is a way to upvote this issue. I am a big fan of smart shots and it was the primary reason I purchased the Solo over other drones. I'm definitely looking forward to authoring my own smart shots if there was more documentation. 

I'm also curious about @franknitty69's question about running custom smartshots via the solo app or if  they can only be invoked via ssh or via a custom dronekit android/ios app?

An additional thought is to have a "smart shot gallery" where we could trade smart shots with other Solo users.

Thanks!