3drobotics/solodevguide #231: Error recording h264 video

franknitty69, 2015-12-05:
in section 2.4, Video Pipeline, an error occurs when recording h264 video.

**Steps to reproduce**
1. in pc terminal, solo video acquire
2. in solo terminal, smart install gst-plugins-bad-videoparsersbad gst-plugins-good-isomp4
3. in solo terminal, gst-launch -e tvsrc device=/dev/video0 ! vpuenc codec=6 bitrate=5000000 ! h264parse ! mp4mux ! filesink location=video.mp4

**Actual Result***
MFW_GST_tvsrc_PLUGIN 3.0.11 build on Aug  6 2015 11:55:52.
Setting pipeline to PAUSED ...
[INFO]	Product Info: i.MX6Q/D/S
vpuenc versions :)
	plugin: 3.0.11
	wrapper: 1.0.46(VPUWRAPPER_ARM_LINUX Build on Aug 19 2015 13:37:13)
	vpulib: 5.4.23
	firmware: 3.1.1.46056
ERROR: Pipeline doesn't want to pause.
Setting pipeline to NULL ...
Freeing pipeline ...

**Expected Result**
gst-launch continures to run until you terminte it by typing Ctrl+C

robinabraham, 2015-12-10:
Hitting the same issue. Any recommendations/fixes? Thanks.

franknitty69, 2016-01-15:
i had an error with the jpg version of this as well so i tried a few things. if you issue the command **modprobe mxc_v4l2_capture** before the **gst-launch** command then it works.

hamishwillee, 2016-02-11:
We're going to do some work to support this use case properly with public APIs. In the meantime we've decided to "unpublish" the current solution (#281). I am therefore closing this issue.

closed 2016-02-11 by hamishwillee