3drobotics/solodevguide #232: Error capturing a JPEG image

franknitty69, 2015-12-05:
in section 2.4, Video Pipeline, an error occurs when capturing a JPEG image

**Steps to reproduce**
1. in pc terminal, solo video acquire
2. in solo terminal, smart install gst-plugins-good-jpeg
3. in solo terminal, gst-launch tvsrc device=/dev/video0 num-buffers=1 ! mfw_ipucsc ! jpegenc ! filesink location=image.jpg

**Actual Result**
MFW_GST_tvsrc_PLUGIN 3.0.11 build on Aug  6 2015 11:55:52.
IPU_CSC_CORE_LIBRARY_VERSION_INFOR_01.00.
MFW_GST_IPU_CSC_PLUGIN 3.0.11 build on Nov 11 2015 22:26:10.
Setting pipeline to PAUSED ...
ERROR: Pipeline doesn't want to pause.
Setting pipeline to NULL ...
Freeing pipeline ...

**Expected Result**
gst-launch runs, takes a picture and then exits gracefully.

leeyh005, 2016-01-13:
Hi,
I have faced same issue. So, I have dug into gstream a bit. 
And I found that after 'solo video acquire' command, 'tvsrc device=/dev/video0' is not available anymore.( But not sure.. I am beginner of gstreamer )
So, I modified like this "gst-launch mfw_v4lsrc num-buffers=1 ! mfw_ipucsc ! jpegenc ! filesink location=image.jpg". 
mfw_v4lsrc is camera. you can check with 'gst-inspect' command. 
With above command, I can get jpg image from gropro.
But, still not sure this is correct method or not.
Please try and give any idea.

franknitty69, 2016-01-15:
i tried the command:

gst-launch mfw_v4lsrc num-buffers=1 ! mfw_ipucsc ! jpegenc ! filesink location=image.jpg

and i get the same error:

MFW_GST_V4LSRC_PLUGIN 3.0.11 build on Nov 11 2015 22:25:56.
IPU_CSC_CORE_LIBRARY_VERSION_INFOR_01.00.
MFW_GST_IPU_CSC_PLUGIN 3.0.11 build on Nov 11 2015 22:26:10.
Setting pipeline to PAUSED ...
ERROR: Pipeline doesn't want to pause.
Setting pipeline to NULL ...
Freeing pipeline ...

franknitty69, 2016-01-15:
so i spent an hour just trying every **gst-launch** command i could come up with. i randomly tried **modprobe mxc_v4l2_capture** and then **gst-launch tvsrc** and my **ERROR: Pipeline doesn't want to pause error** went away.

so i tried the full command **gst-launch tvsrc device=/dev/video0 num-buffers=1 ! mfw_ipucsc ! jpegenc ! filesink location=image.jpg** and it worked.

leeyh005, 2016-01-15:
True.
The best way to resolve this issue is 'modprove mxc_v4l2_capture' again.
thank you

cheers

GPSFan, 2016-02-06:
When you run "solo video acquire", the /usr/bin/video_send.sh file is modified to add a test for the file /root/home/ACQUIRE_VIDEO_FEED. 
If that file is present, video_send.sh doesn't modprobe mxc_v4l2_capture. (or do anything else for that matter)
If that file is not present, video_send.sh operates normally and modprobes the capture driver, starts sn_master_snd as well as sn_encoder. If any of the above crashes or stops, they are restarted.
You need the capture driver for the video pipeline to run, if not, it can't pause because it is not running.
It should be possible to have another video capture driver running instead of mxc_v4l2_capture so long as it creates /dev/video0. Maybe even v4l2loopback as it creates /dev/videox. But someone from 3dr would have to supply the v4l2loopback.ko since the one in the repository was compiled for an older kernel version and won't load with the current kernel. See Issue #256.

hamishwillee, 2016-02-11:
We're going to do some work to support this use case properly with public APIs. In the meantime we've decided to "unpublish" the current solution (#281). I am therefore closing this issue.

closed 2016-02-11 by hamishwillee

alexblack, 2016-03-24:
Any update here? I am struggling to get a video frame captured in any manner whatsoever, eg python app, or gst-launch, etc.

alexblack, 2016-03-24:
Through some black magic steps I managed to get to a state where I can capture a frame! But, opencv in python still refuses to open the camera.

```
root@3dr_solo:/opt/ddc# gst-launch mfw_v4lsrc num-buffers=1 ! mfw_ipucsc ! jpegenc ! filesink location=image.jpg
MFW_GST_V4LSRC_PLUGIN 3.0.11 build on Mar  4 2016 02:10:26.
IPU_CSC_CORE_LIBRARY_VERSION_INFOR_01.00.
MFW_GST_IPU_CSC_PLUGIN 3.0.11 build on Mar  4 2016 02:10:41.
Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
Got EOS from element "pipeline0".
Execution ended after 913616000 ns.
Setting pipeline to PAUSED ...
Setting pipeline to READY ...
Setting pipeline to NULL ...
Freeing pipeline ...
```

franknitty69, 2016-03-24:
Alex there a few different photo/video issues in this repo but they are all related (see #256). i don't think there is anything that can be done until 3dr releases a better way to handle video pragmatically.

alexblack, 2016-03-24:
I managed to get things working using SoloCamera found here: https://github.com/3drobotics/solodevguide/tree/b978c602b6cf20e82b454770584e937ee3b017da/examples/opencv