3drobotics/solodevguide #247: droneapi/dronekit installation to solo error

mikkab, 2015-12-28:
I tying to install dronekit or droneapi via pip to solo as described http://dev.3dr.com/starting-installing.html but receive error "unable to execute arm-oe-linux-gnueabi-gcc: No such file or directory"
What am I doing wrong? Where I can get gcc/arm-oe-linux-gnueabi-gcc for compilation?

sample result:
pip install droneapi
Collecting droneapi
/drone/env/lib/python2.7/site-packages/pip/_vendor/requests/packages/urllib3/util/ssl_.py:90: InsecurePlatformWarning: A true SSLContext object is not available. This prevents urllib3 from configuring SSL appropriately and may cause certain SSL connections to fail. For more information, see https://urllib3.readthedocs.org/en/latest/security.html#insecureplatformwarning.
  InsecurePlatformWarning
  Downloading droneapi-1.5.0.tar.gz
Collecting pymavlink>=1.1.50 (from droneapi)
  Using cached pymavlink-1.1.66.tar.gz
Collecting MAVProxy>=1.4.13 (from droneapi)
  Downloading MAVProxy-1.4.38.tar.gz (229kB)
    100% |################################| 229kB 83kB/s 
Collecting protobuf>=2.5.0 (from droneapi)
  Downloading protobuf-2.6.1.tar.gz (188kB)
    100% |################################| 188kB 112kB/s 
Collecting requests==2.5.1 (from droneapi)
  Downloading requests-2.5.1-py2.py3-none-any.whl (464kB)
    100% |################################| 466kB 63kB/s 
Collecting pyserial (from MAVProxy>=1.4.13->droneapi)
  Downloading pyserial-2.7.tar.gz (122kB)
    100% |################################| 122kB 217kB/s 
Requirement already satisfied (use --upgrade to upgrade): setuptools in ./env/lib/python2.7/site-packages (from protobuf>=2.5.0->droneapi)
Building wheels for collected packages: droneapi, pymavlink, MAVProxy, protobuf, pyserial
  Running setup.py bdist_wheel for droneapi
  Stored in directory: /home/root/.cache/pip/wheels/40/8d/77/607555a6569a5804a880c30be9d84ef947d04d0680a20f2dc9
  Running setup.py bdist_wheel for pymavlink
  Complete output from command /drone/env/bin/python -c "import setuptools;__file__='/var/volatile/tmp/pip-build-uW9OZf/pymavlink/setup.py';exec(compile(open(__file__).read().replace('\r\n', '\n'), __file__, 'exec'))" bdist_wheel -d /tmp/tmpDia61npip-wheel-:
  running bdist_wheel
  running build
  running build_py
  creating build
  creating build/lib.linux-armv7l-2.7
  creating build/lib.linux-armv7l-2.7/pymavlink
  copying ./rotmat.py -> build/lib.linux-armv7l-2.7/pymavlink
  copying ./mavparm.py -> build/lib.linux-armv7l-2.7/pymavlink
  copying ./setup.py -> build/lib.linux-armv7l-2.7/pymavlink
  copying ./mavlinkv10.py -> build/lib.linux-armv7l-2.7/pymavlink
  copying ./mavutil.py -> build/lib.linux-armv7l-2.7/pymavlink
  copying ./mavwp.py -> build/lib.linux-armv7l-2.7/pymavlink
  copying ./__init__.py -> build/lib.linux-armv7l-2.7/pymavlink
  copying ./mavextra.py -> build/lib.linux-armv7l-2.7/pymavlink
  copying ./DFReader.py -> build/lib.linux-armv7l-2.7/pymavlink
  copying ./fgFDM.py -> build/lib.linux-armv7l-2.7/pymavlink
  copying ./mavlinkv09.py -> build/lib.linux-armv7l-2.7/pymavlink
  copying ./a.py -> build/lib.linux-armv7l-2.7/pymavlink
  copying ./quaternion.py -> build/lib.linux-armv7l-2.7/pymavlink
  creating build/lib.linux-armv7l-2.7/pymavlink/generator
  copying ./generator/mavparse.py -> build/lib.linux-armv7l-2.7/pymavlink/generator
  copying ./generator/mavcrc.py -> build/lib.linux-armv7l-2.7/pymavlink/generator
  copying ./generator/mavgen_objc.py -> build/lib.linux-armv7l-2.7/pymavlink/generator
  copying ./generator/mavtestgen.py -> build/lib.linux-armv7l-2.7/pymavlink/generator
  copying ./generator/mavgen_javascript.py -> build/lib.linux-armv7l-2.7/pymavlink/generator
  copying ./generator/mavlink.py -> build/lib.linux-armv7l-2.7/pymavlink/generator
  copying ./generator/__init__.py -> build/lib.linux-armv7l-2.7/pymavlink/generator
  copying ./generator/mavgen_swift.py -> build/lib.linux-armv7l-2.7/pymavlink/generator
  copying ./generator/mavtemplate.py -> build/lib.linux-armv7l-2.7/pymavlink/generator
  copying ./generator/mavgen_wlua.py -> build/lib.linux-armv7l-2.7/pymavlink/generator
  copying ./generator/mavgen_cs.py -> build/lib.linux-armv7l-2.7/pymavlink/generator
  copying ./generator/mavgen_c.py -> build/lib.linux-armv7l-2.7/pymavlink/generator
  copying ./generator/mavgen_python.py -> build/lib.linux-armv7l-2.7/pymavlink/generator
  copying ./generator/mavgen.py -> build/lib.linux-armv7l-2.7/pymavlink/generator
  copying ./generator/mavgen_java.py -> build/lib.linux-armv7l-2.7/pymavlink/generator
  creating build/lib.linux-armv7l-2.7/pymavlink/generator/lib
  copying ./generator/lib/__init__.py -> build/lib.linux-armv7l-2.7/pymavlink/generator/lib
  creating build/lib.linux-armv7l-2.7/pymavlink/generator/lib/genxmlif
  copying ./generator/lib/genxmlif/xmlifDom.py -> build/lib.linux-armv7l-2.7/pymavlink/generator/lib/genxmlif
  copying ./generator/lib/genxmlif/xmlifODict.py -> build/lib.linux-armv7l-2.7/pymavlink/generator/lib/genxmlif
  copying ./generator/lib/genxmlif/__init__.py -> build/lib.linux-armv7l-2.7/pymavlink/generator/lib/genxmlif
[... 96 lines cut ...]
  
  ----------------------------------------
  Failed building wheel for pymavlink
  Running setup.py bdist_wheel for MAVProxy
  Stored in directory: /home/root/.cache/pip/wheels/4a/18/3c/3a465ac974c68258b813bb6d4744239f79487047bec3ea15aa
  Running setup.py bdist_wheel for protobuf
  Stored in directory: /home/root/.cache/pip/wheels/2c/40/46/9dc67b28cbea4b3c9752dde3abd4787d96c318a5292af8c1de
  Running setup.py bdist_wheel for pyserial
  Stored in directory: /home/root/.cache/pip/wheels/46/06/d0/0a8f8136db90567df3ed02d9d15391178e350576a7a7fb03fa
Successfully built droneapi MAVProxy protobuf pyserial
Failed to build pymavlink
Installing collected packages: pymavlink, pyserial, MAVProxy, protobuf, requests, droneapi
  Running setup.py install for pymavlink
    Complete output from command /drone/env/bin/python -c "import setuptools, tokenize;__file__='/var/volatile/tmp/pip-build-uW9OZf/pymavlink/setup.py';exec(compile(getattr(tokenize, 'open', open)(__file__).read().replace('\r\n', '\n'), __file__, 'exec'))" install --record /tmp/pip-HuBmRL-record/install-record.txt --single-version-externally-managed --compile --install-headers /drone/env/include/site/python2.7/pymavlink:
    running install
    running build
    running build_py
    running build_ext
    building 'mavnative' extension
    arm-oe-linux-gnueabi-gcc -march=armv7-a -mthumb-interwork -mfloat-abi=hard -mfpu=neon -mtune=cortex-a9 -D__SOFTFP__ -fno-strict-aliasing -O2 -pipe -g -feliminate-unused-debug-types -DNDEBUG -g -O3 -Wall -Wstrict-prototypes -fPIC -Igenerator/C/include_v1.0 -Imavnative -I/usr/include/python2.7 -c mavnative/mavnative.c -o build/temp.linux-armv7l-2.7/mavnative/mavnative.o
    unable to execute arm-oe-linux-gnueabi-gcc: No such file or directory
    error: command 'arm-oe-linux-gnueabi-gcc' failed with exit status 1
    
    ----------------------------------------
Command "/drone/env/bin/python -c "import setuptools, tokenize;__file__='/var/volatile/tmp/pip-build-uW9OZf/pymavlink/setup.py';exec(compile(getattr(tokenize, 'open', open)(__file__).read().replace('\r\n', '\n'), __file__, 'exec'))" install --record /tmp/pip-HuBmRL-record/install-record.txt --single-version-externally-managed --compile --install-headers /drone/env/include/site/python2.7/pymavlink" failed with error code 1 in /var/volatile/tmp/pip-build-uW9OZf/pymavlink

hamishwillee, 2015-12-28:
@mikkab We no longer use droneapi (it recently updated to dronekit) - I will submit a patch to fix that document. We do have better instructions but they don't appear to have been built to the live site. While I get that sorted out, you can see the documents you need here:
https://github.com/3drobotics/solodevguide/blob/master/book/concept-dronekit.md
https://github.com/3drobotics/solodevguide/blob/master/book/example-helloworld.md

hamishwillee, 2015-12-28:
@mrpollo Can you please merge #248 to fix the defect outlined here? Then can you please ensure that the new version is rebuilt (the master contains a lot of content that is not being published).

mikkab, 2015-12-28:
Thanks, there is a new way. Installation via solo script pack/run works for whl files, but running "import dronekit" in python shows following error: ERROR LOADING MAVNATIVE - falling back to python implementation
and connection to solo localhost is failed by timeout
vehicle = dronekit.connect('127.0.0.1:14550', wait_ready=True)
...
  File "<stdin>", line 1, in <module>
  File "/log/solo-script/env/lib/python2.7/site-packages/dronekit/__init__.py", line 2354, in connect
    vehicle.initialize(rate=rate, heartbeat_timeout=heartbeat_timeout)
  File "/log/solo-script/env/lib/python2.7/site-packages/dronekit/__init__.py", line 1815, in initialize
    raise APIException('Timeout in initializing connection.')
dronekit.APIException: Timeout in initializing connection.

If I trying to install latest dronekit via pip, problem is the same as droneapi - arm-oe-linux-gnueabi-gcc is not found.

mikkab, 2015-12-28:
'udpin:0.0.0.0:14550' works fine

hamishwillee, 2015-12-28:
Yes, `udpin:0.0.0.0:14550` is the correct endpoint to use for Solo.

josealb, 2017-02-06:
I am having the same problem with "error: command 'arm-oe-linux-gnueabi-gcc' failed with exit status 1"
When installing pymavlink
I can't install any of the newer versions on the Solo
Am I doing something wrong?

peterbarker, 2017-02-07:
On Mon, 6 Feb 2017, josealb wrote:

> I am having the same problem with "error: command 'arm-oe-linux-gnueabi-gcc'
> failed with exit status 1"
> When installing pymavlink
> I can't install any of the newer versions on the Solo
> Am I doing something wrong?

No.  We've had a bit of an oops with pymavlink, and the "fix" we have in 
place requires a C compiler to be installed.

If anyone out there has a C compiler that can be installed on Solo... 
well, that would be appreciated....

Additionally, if anyone can find a 2.0.6 tarball of pymavlnk from a week 
ago, that would be more useful....

alexblack, 2017-02-07:
I posted instructions last May on how to install gcc, does that help @peterbarker ?

https://github.com/3drobotics/solodevguide/issues/255#issuecomment-217662999

peterbarker, 2017-02-07:
On Mon, 6 Feb 2017, Alex Black wrote:

> I posted instructions last May on how to install gcc, does that help @peterbarker ?
> 
> #255 (comment)

Yes, yes it does!

Could someone who here who has had the issue attempt to install gcc per 
Alex's instructions and report back as to whether it fixes the pymavlink 
issue, please?

Peter

jimmy1409, 2017-05-30:
Hi @alexblack  Can you upload the command which can install the GCC compiler in solo?

alexblack, 2017-05-30:
hi @jimmy1409 its posted here: https://github.com/3drobotics/solodevguide/issues/255#issuecomment-217662999

jimmy1409, 2017-06-02:
Hi guys,

When I try to run "pip install pymavlink-2.0.8" on my "Solo" , I am getting this error.
I don't have apt-get or yum in my solo.

I am getting this error.

   creating build/temp.linux-armv7l-2.7/mavnative
    arm-oe-linux-gnueabi-gcc -march=armv7-a -mthumb-interwork -mfloat-abi=hard -mfpu=neon -mtune=cortex-a9 -D__SOFTFP__ -fno-strict-aliasing -O2 -pipe -g -feliminate-unused-debug-types -DNDEBUG -g -O3 -Wall -Wstrict-prototypes -fPIC -Igenerator/C/include_v1.0 -Igenerator/C/include_v2.0 -Imavnative -I/usr/include/python2.7 -c mavnative/mavnative.c -o build/temp.linux-armv7l-2.7/mavnative/mavnative.o
    unable to execute arm-oe-linux-gnueabi-gcc: No such file or directory
    error: command 'arm-oe-linux-gnueabi-gcc' failed with exit status 1
    
    ----------------------------------------
Command "/home/root/env/bin/python -u -c "import setuptools, tokenize;__file__='/var/volatile/tmp/pip-build-iz7CaI/pymavlink/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-qukLR8-record/install-record.txt --single-version-externally-managed --compile --install-headers /home/root/env/include/site/python2.7/pymavlink" failed with error code 1 in /var/volatile/tmp/pip-build-iz7CaI/pymavlink/

How can I solve this error?
Can I install apt-get in my solo? If I can , then how?

Thanks,

jimmy

jimmy1409, 2017-06-04:
Hi @alexblack  , Thank you for your quick response. but I am not able to install C compiler in my solo. 

sudo apt-get install gcc-arm-linux-gnueabi

I think this is the solution for this whole problem. If I can install gcc-arm-linux-gnueabi then, arm will be able to execute C file. 

But there is main problem in order to install this thing. Solo doesn't have apt-get, so I can't use apt-get. I am using "smart" in solo. But "smart" repository doesn't have this library. Can anyone help me out here that how can I install this library in my Solo?

I am 99% sure , it will work after installing this library.

alexblack, 2017-06-04:
Hi @jimmy1409 did you try the steps I used here: https://github.com/3drobotics/solodevguide/issues/255#issuecomment-217662999