3drobotics/solodevguide #251: Sidebar is broken in local builds

hamishwillee, 2016-01-06:
This is caused by #210 - changelist: https://github.com/3drobotics/solodevguide/commit/3038bf06264c94943a6b622e5ebb9f05991fe23a

Basically on a local build the .js files don't live under a website directory.

@mrpollo Can we get this fixed in a way that works for website and local builds. It makes it very hard to visualise site structure when developing it.

hamishwillee, 2016-01-20:
@mrpollo Please look at this. Complete pain for me.

mrpollo, 2016-01-26:
Can you show me your lib versions?

```
node --version
npm --version
gitbook --version
```

hamishwillee, 2016-01-26:
I don't think this has anything to do with my versions. This fails because the /website folder does not exist on local builds. If I revert to before the changelist before you added this path (see first post in this issue) everything works.

For completeness
```

D:\Github\solodevguide>node --version
v5.2.0

D:\Github\solodevguide>npm --version
3.3.12

D:\Github\solodevguide>gitbook --version
1.0.1
```