3drobotics/solodevguide #252: Architectural diagram clarifications and updates

hamishwillee, 2016-01-06:
The diagram in the  [Architectural Overview](http://dev.3dr.com/concept-architecture.html) is incomplete (missing Accessory bay) and also in some cases a little confusing. We should tidy it. Would appreciate answers to the following questions. 

Rough update (for discussion only) below. New lines shown in red, new box in blue. 

[image]

1. Where is the original source for this diagram (I can reproduce existing version if needed, but would be faster if we had original)
2. What do the direction of the arrows mean? I am assuming they mean "direction of messages"? So the HDMI arrow from gopro indicates that images are sent from gopro to companion computer?
   - Are all the other arrows correct? 
     - Is LED really sending messages to Pixhawk?
     - I think  Pixhawk to Gimbal is only one way - we use this to send commands to control gimbal angle and take pics. Correct?
     - I've drawn Pixhawk to GoPro as 2 way, because I think this also controls settings.
     - USB between Pix2 and Companion computer is shown as one way - That looks wrong ???
     - USB companion computer to accessory bay is 2 way too right?
3. Are there any lines "not available to developers" (I could show these dotted). For example, I believe that the USB2 line between the Gopro and SoloLink is not available to developers. Correct? Any more?
4. What is the "Pair" about?
5. Accessory bay. Please check my new box. Are the directions of the arrows OK? Are there any missing lines?
6. Battery bay. Is this correct - just a box with I2C line pointing from battery to pixhawk? Anything else useful to add here?

Are there any other components that should be on the diagram?

hamishwillee, 2016-01-06:
@dmoroni @tcr3dr Any help/information you can give on this would be great. I'm trying to have a "master" diagram and then I'll create a sub diagram for individual sections (e.g. accessory bay)

dmoroni, 2016-01-21:
1. I am not sure where the original source of this diagram is, but I have been using google drawings and have a similar diagram [here] (https://drive.google.com/open?id=12sN11k6beLbw1Vs7vPkQuf5n-t_qernkAT_uHyUnuZY). We should probably all use the same tool.

2. I think the arrows are meant to be directional (Host -> Device or Source -> receiver) but they don't all seem correct. 
+ The battery bay is bidirectional (I think)
+ the lights don't communicate to pixhawk (I think)...
+ The USB I think is meant to show which is host and which is the device but thats nto clear

3. I think you are incorrect, the usb2 line between the Gopro and Solo is in the gimbal bay. Actually this diagram is not very clear and should show where these ports and bays

4. I think pair is the button on the bottom

5. Looks right to me, we should add a line to the battery

6. battery bay looks right, but we should also show how the accessory bay and gimbal bay have unregulated battery power lines

7. no idea

timlee3dr, 2016-01-22:
>    3. The battery bay is bidirectional (I think)
>    4. the lights don't communicate to pixhawk (I think)...

2. Arrows direction of communication. bidirectional is arrows on both ends
- There is a MCU between the LED and Pixhawk on the ESC and it is capable
of bidirectional communication (I2C). Current only command acknowledgements
come back to the pixhawk. But to make is simple the arrow should only go
from Pixhawk to LEDs.
- The gimbal communication path supports bidirectional serial. Not sure if
FW does the communication in both ways. I don't think this is MAVLINK tho.
- A USB mux is between the Pixhawk and Sololink & Gopro. The USB connection
can switch back between the Pixhawk <--> Gopro or Pixhawk <--> Sololink
- USB connection between Sololink and Accessory bay is directional. USB is
a bidirectional link.
4. There a pair button that needs to be pressed to start the pairing
process between the controller and Solo
5. Bidirectional is correct
6. I2C is bidirectional. I don't think there's anything else.

3.

hamishwillee, 2016-01-22:
Thanks very much @dmoroni @timlee3dr 

Dan, I've recreated as a Lucid diagram in your 3DR SoloDevGuide/Source Files directory - and shared with you and Tim. Block diagramming with Google tools has advantage it is available to everyone, but Lucid makes it much easier to produce great looking diagrams. 

I think I've captured everything. Can you guys please confirm the image snapshot below is OK (?) and see my notes below it.

[image]

1. @timlee3dr I didn't include the MUX. Is this "useful" for accessory or internal developers to know about? Ie do they need to do anything "special" because of this implementation or is it transparent to them.
2. Dan, Tim - "accessory bay and gimbal bay have unregulated battery power lines"
  * Is there an electrical symbol or other way of showing this?

timlee3dr, 2016-01-22:
The mux is necessary As the current diagram shows 1 extra USB connection in
the that the sololink does not support.

Adding a red connection between the battery, accessory, and gimbal can show
this. A legend can be added to show what the lines recommend. Also I'll
share a 3DR internal detailed connection diagram for your reference.

kevinmehall, 2016-01-22:
Our accessory management code will have to control the mux to detect devices on that port, so the only impact on accessory developers is that the accessory will disconnect during Pixhawk updates. But for internal use, and before the accessory API is developed, it would be good to document somewhere.

If you want to account for every board and cable in the vehicle, you're missing a few components:
  * ESCs, connected via PWM from the Pixhawk, which could share a box with the LEDs (the arrow goes the other way)
  * GPS, connected via UART to the Pixhawk
  * Compass, connected via I2C to the PIxhawk

hamishwillee, 2016-01-22:
@kevinmehall Thanks very much for that. My understanding is that we don't need to cover GPS, Compass or ESC, as we haven't yet developed an accessory story/plan for those. Until we do, I'd prefer to keep the diagram relatively simple. Will confirm with Dan.

With respect to the MUX, I have added this to the diagram here: https://www.lucidchart.com/publicSegments/view/3ab77e97-c6a1-4230-8f95-064ae3002634/image.png

hamishwillee, 2016-02-10:
This has been fixed by #260

closed 2016-02-10 by hamishwillee