3drobotics/solodevguide #253: Gimbal Bay is missing hardware spec + some questions

hamishwillee, 2016-01-07:
[Gimbal Bay](http://dev.3dr.com/hardware-gimbalbay.html) Questions:

Overview

* Are camera/gimbal solutions the **only** thing we're recommending for this bay? For example, are we precluding non-camera solutions like a big fat powerful flashlight instead. Currently we have:

  > The Gimbal Bay is an extensible interface to which any “Made For Solo”-approved 3rd party vendors can sell an attachable 3-axis gimbal and camera solution. 

Electrical Interface section:

* Add text marked in bold:

  > The board side connector **for the gimbal cable** can be found on ...

* Can you do anything with the HDMI cable as a developer? If not (as you'd expect), perhaps add a generic statement that its a standard HDMI cable and there are no developer opportunities?

Hardware spec

* This section is missing. 

   * Do we have an ETA or a plan for an ETA?
   * What is required? I am guessing 
     * a schematic showing the size of the "hole", the screw positions, and a screw length, description of the clip size/shape? 
     * A CAD model people can use as the base design for their own covers
     * A diagram showing the available dimensions of the space within Solo.
     * A diagram showing the maximum/recommended radius/space outside of Solo, in order for the gimbal to not impact on accessory bay.
     * What else?

Developer spec (section)

* Should describe the contents of the section, not act as an introduction to just the Gimbal cable:

  > The Gimbal Cable has three primary responsibilities

  I think that this text and the following three headings should be pushed down a level and grouped under a heading "Gimbal Cable" OR you should remove this first line, and put "Gimbal Cable" in the following headings. 

Co-processing with USB

* This text should be made into a note or warning:

  > Communication between the Solo co-processor and the gimbal is not yet available for 3rd party developers.

Software Interface

* This section is a complete mess. From this I can't tell what a user can actually do on Solo to integrate with a camera. My assumption based on this information though is "absolutely nothing" unless they can force their device to look like a gopro.

* The introduction is very confusing! There are lots of mixed messages. Essentially we need to know what you can actually do/not do to integrate with Solo. 

  Things that confused me ...

  >  The custom Solo Gimbal protocol is not yet supported by any released versions of Ardupilot.

  How then does ArduPilot ever control the camera? I think what you are saying is that Solo uses a custom camera protocol and hence a custom ArduPilot? If so, what implications does that have for users?

  > Ardupilot already has the ability to communicate with several types of pre-existing gimbal controllers including SToRM32 and Alexmos. 

  So does that mean that if I shove Storm gimbal on everything will work seamlessly? I don't see how that can be true since Solo is sending the Gopro protocol.

  > For development purposes, a Pixhawk 1 and Copter 3.3 Ardupilot firmware are recommended.   The Solo development stack is not yet ready to integrate multiple types of gimbals so it will not work on Solo.

    What? I think you're saying that actually, there is no way to integrate with the gimbal!

Common Gimbal Camera Control

* Basically this says that you can't actually intercept these commands or work with the camera? What needs to happen before it will be supported?

Custom Gimbal Camera Control

* Defines what a mavlink command long looks like and explains that you can create your own commands or use the existing ones. What is the purpose of this? I don't see what you're trying to tell developers when they have no access to control the messages etc.

hamishwillee, 2016-01-07:
@dmoroni @tcr3dr Some questions about the gimbal bay doc. Would appreciate your feedback. The software story, in particular, is very unclear.

dmoroni, 2016-01-21:
I will try to capture answers. We should discuss in a call to help clear this up.

1. Yes this bay is not only used for cameras, but it is meant for cameras and where you would put a camera

2. No you cannot do anything with the HDMI cable as a developer other than shove video into it

3. Agree we are missing hardware info. Can you create a checklist of content you need and we can assign owners to creating it?

4. The software integration story is not complete, I agree if we don't have it maybe it shouldn't be populated

Focus on describing the architecture and hardware (mech / elec) interfaces and specs. We will fill in the software story later.

hamishwillee, 2016-01-28:
Thanks @dmoroni 

>1. Yes this bay is not only used for cameras, but it is meant for cameras and where you would put a camera

I have simplified the introduction, linking to MFS and adding a note to say that other accessories are supported (this will go in with #266):

[image]

> 4. The software integration story is not complete, I agree if we don't have it maybe it shouldn't be populated

For the moment I have moved some of the "dev story" sections under the gimbal cable and added the following note at the top of the software section. Basically it says "this is what you can do right now, and its basically play on raw ardupilot"
OK solution for you?

[image]

> 3. Agree we are missing hardware info. Can you create a checklist of content you need and we can assign owners to creating it?

The main things we are missing are:
* A description/mechanical diagram of the "base plate" for attaching the gimbal to Solo. This needs to cover the interface points:

   * Shape of a "bay cover" including curvature
   * Shape of the clips up the front
   * Position, size and any other information for the four screws
   * INfomration about go-no go areas under the plate - for example clearance needed around screws.
   
I think the best way to provide this would probably be a CAD diagram of a "cover" for the solo to replace any real gimbal (like the battery CAD file). It would be such that people could come along and modify it for their own purposes (e.g. it might have the biggest possible hole for gimbal to poke through, and they could reduce it).

* A description of the space avaialble inside the body of the bay - again, enough for people to work out what solutions are basically feasible.

* A close up of the keepout areas for the gimbal based on the existing gimbal.

Anything else?

hamishwillee, 2016-01-29:
@dmoroni Thanks for the CAD file. I've addeed mechanical section as part of #266. 

We still need screw, keepout, and infomration about the inside cavity of the solo, but this is much better.