3drobotics/solodevguide #255: libsolocam.so supports only 720p grayscale

sjelavic, 2016-01-14:
Looks like this lib supports only one video format. Any plan of expanding the capability. I use it with face detection and would like to see the fps figures for other video formats as well, including color.

kevinmehall, 2016-01-14:
The [buffer returned from the library](https://github.com/3drobotics/solodevguide/blob/115fdafb27cafd2640dfc23b7e0f6806d3efdbbc/examples/opencv/SoloCamera.py#L66) is actually a [YUV420](https://en.wikipedia.org/wiki/YUV#Y.27UV420p_.28and_Y.27V12_or_YV12.29_to_RGB888_conversion) image. The python code only uses the Y plane to get a grayscale image, but you could modify it to use the color data in the U and V planes as well.

Once you `solo video acquire`, the HDMI input is just a standard V4L2 device at /dev/video0, albeit without the ability to set modes or resolutions, since it gets whatever the GoPro is sending.

The `solocam` library is just a V4L2 wrapper intended as a more convenient and efficient way of getting grayscale images. Using the [default gstreamer pipeline](http://dev.3dr.com/concept-video.html), we were converting from YUV -> RGB -> grayscale, which is inefficient when the hardware effectively already provides grayscale in the form of the Y plane in YUV. [Here's the source to libsolocam.so](https://gist.github.com/kevinmehall/17f2f1ad113f4b4349dc) if you want to see what it's doing and/or modify it.

sjelavic, 2016-01-14:
Thanks for the input @kevinmehall . That should work for me.

OXINARF, 2016-01-14:
@kevinmehall The source for libsolocam.so should be in this repo, otherwise it's going to get lost in this issue

sjelavic, 2016-01-19:
Have a few more questions Kevin,
Which cross compiler do you recommend to compile the library.
I have cygwin on one machine and umbutu on another one.

Also how do you modify solocam.c
Do you just change the pixelformat to V4L2_PIX_FMT_RGB or something like
that?

Thanks,
Stan

DarylWM, 2016-03-24:
Hi @kevinmehall - thanks for your work on the library. I'm keen to get colour images as well, so would you be able to share some instructions for building it too?

kevinmehall, 2016-03-24:
The image from libsolocam.so is actually a color image in YUV420 format, despite the function name. This is what comes off the HDMI port, and is the only mode you'll get (at least with the GoPro HDMI input I've tested). It just happens that a YUV420 image begins with the Y plane, which is a grayscale version of the image, followed by color data in the U and V planes. The original application was some computer vision code that needed grayscale, so this was convenient because it could just ignore the color planes when it [converts the Y plane to a numpy array](https://github.com/3drobotics/solodevguide/blob/master/examples/opencv/SoloCamera.py#L67).

You can't change the pixel format at the V4L2 level -- you have to [convert it yourself](https://en.wikipedia.org/wiki/YUV#Y.27UV420p_.28and_Y.27V12_or_YV12.29_to_RGB888_conversion) or use a library like gstreamer to do it for you.

Compiling C code for Solo is kind of a big missing piece in this dev guide, in part because the toolchain that 3DR developers have handy comes from the Yocto-based tree we use to build the entire SoloLink system internally. Let me try and see if there's an easy solution for this case, though...

kevinmehall, 2016-03-24:
This builds it on my Ubuntu 15.10 system, but I haven't yet tested the resulting binary on a Solo to make sure it links properly:

```
sudo apt-get install gcc-arm-linux-gnueabi
arm-linux-gnueabi-gcc solocam.c -O3 -lm -shared -o libsolocam.so
```

sjelavic, 2016-03-28:
Thanks @kevinmehall I will try the build. In the meantime I tried using the YUV420 buffer but had a hard time getting the color frame in python. Ended up using the following for color frame:
   os.system("modprobe mxc_v4l2_capture")
    os.system("gst-launch -e tvsrc device=/dev/video0 num-buffers=1 !  mfw_ipucsc ! jpegenc ! filesink location=cam.jpg")
    frame = cv2.imread('cam.jpg',1)

This worked but it is pretty slow. So maybe I should have another look at the buffer returned from libsolocam.so and see if I can compose the color frame from it and then if that does not work modify and rebuild the lib.

alexblack, 2016-03-31:
Anyone had any luck compiling `solocam.c` to run on the solo?

paulovsm, 2016-04-10:
I could compile solocam but I still can't run SoloCamera example because of `wrong ELF class: ELFCLASS32` error.

alexblack, 2016-04-10:
Cool, how did you compile it? Did you compile it on the solo itself?

alexblack, 2016-05-07:
I managed to get it to compile, next step - try it out.

```
smart channel --remove cortexa9hf_vfp_neon
smart channel --add cortexa9hf_vfp_neon type=rpm-md baseurl=http://downloads.yoctoproject.org/releases/yocto/yocto-1.5.1/rpm/cortexa9hf_vfp_neon/ -y
smart channel --add cortexa9hf_vfp_neon_mx6 type=rpm-md baseurl=http://downloads.yoctoproject.org/releases/yocto/yocto-1.5.1/rpm/cortexa9hf_vfp_neon_mx6/ -y
smart update
smart install gcc gcc-symlinks -y
smart install libc6-dev gcc-dev binutils -y
gcc -v

gcc --std=c99 -c -Wall -Werror -fpic solocam.c
gcc -shared -o libsolocam.so solocam.o
```

alexblack, 2016-05-17:
I've been able to modify, compile and use new `libsolocam.so` builds no problem.

hamishwillee, 2016-05-17:
@alexblack Very cool. If you're so inclined you might want to create and point to instructions on how to do this ... perhaps in the wiki.

alexblack, 2016-05-17:
hey @hamishwillee I instructions just above!

hamishwillee, 2016-05-17:
@alexblack I am blind. Thank you!

alexblack, 2016-05-29:
Anyone managed to capture video at 1920x1080? See https://github.com/3drobotics/solodevguide/issues/313

alexblack, 2016-05-31:
I think I see how to get color images!  I haven't tried it. But this line:

https://github.com/3drobotics/solodevguide/blob/master/examples/opencv/SoloCamera.py#L67

```
image = np.ctypeslib.as_array(bufc.data,shape = (self.height*self.width,)).reshape(self.height, self.width)
```

Is what converts the color YUV420 image into grayscale.  In [https://en.wikipedia.org/wiki/YUV#Y.27UV420p_.28and_Y.27V12_or_YV12.29_to_RGB888_conversion](YUV420), the Y channel comes before the others, and its luminance.

So that line above creates an array backed by the data starting at the start of the buffer, up to and including the first w*h values, which is all the Y pixels.

If you instead made the array include the U and V channels, then you'd have color, and you could ask opencv to convert it to something sensible perhaps with:

```
img = cv2.cvtColor(img, COLOR_YUV420p2RGB)
```

sjelavic, 2016-05-31:
This is interesting. Will give it a try. I have not used the drone for a while and will be getting back into it.

David-Proconsi, 2016-06-01:
I tried it and it works perfectly:

```
bufp = BUF_P()
check(solocam.solocam_read_frame(ctx, byref(bufp)))
bufc = bufp.contents
imageYUV = np.ctypeslib.as_array(bufc.data,shape = (bufc.used,)).reshape(int(height*1.5), width)
color = cv2.cvtColor(imageYUV,cv2.COLOR_YUV420P2RGB)
```

It is quick and gives you the image in full color.

alexblack, 2016-06-03:
Nice work!  I'd suggest compiling a new `libsolocam` (see above) and change the format to RGB right off the bat, probably does it in hardware and faster..?

ayushajain, 2016-07-19:
can anyone share their compiled libsolocam for rgb?

tasarmiento, 2016-10-24:
Would someone please share their recompiled libsolocam?

tasarmiento, 2017-02-27:
@alexblack @David-Proconsi @sjelavic @kevinmehall 

Did anyone encounter a   "modprobe: FATAL: Module mxc_v4l2_capture not found" error when using a recompiled libsolocam?

The entire output is:

modprobe: FATAL: Module mxc_v4l2_capture not found.
VIDIOC_S_INPUT error 22, Invalid argument
Traceback (most recent call last):
  File "SoloCamera.py", line 81 in <module>
    cam = SoloCamera()
  File "SoloCamera.py", line 29 in __init__
    check(solocam.solocam_open_hdmi(byref(self.ctx)))
  File "SoloCamera.py", line 14 in check
    raise OSError("Error in solace C call (see stderr)")
OSError: Error in solocam C call (see stderr)
Segmentation fault (core dumped)

I am using it on a laptop with Ubuntu 14.04.

tasarmiento, 2017-03-02:
@alexblack @David-Proconsi 
Did you have success in increasing the fps with a recompiled libsolocam?

tasarmiento, 2017-04-07:
@alexblack @David-Proconsi 
In the YUV to RGB conversion did you have issues with lines in the image, or an image that looked stretched (too wide)?

alexblack, 2017-05-30:
@tasarmiento no, I wasn't able to change the FPS or resolution :(