3drobotics/solodevguide #256: v4l2loopback.ko driver version problem

GPSFan, 2016-01-16:
I'll open this as a separate issue since my project is dead in the water without being able to use the v4l2loopback driver.
See issue #213 as well.

From the Solo Development Guide:

"If we wanted to create multiple video processing endpoints, you can activate this driver as follows:
modprobe v4l2loopback exclusive_caps=0,0 video_nr=1,2"

The version 1.3.0 of the solo firmware uses kernel:
"Linux 3dr_solo 3.10.17-rt12-1.0.0_ga+g4176d57 #8 SMP PREEMPT Wed Nov 18 20:27:35 UTC 2015 armv7l GNU/Linux"

After installing solo-cli, running:
"solo install-smart"
Will install smart pointing to the solo repositories at:

PACKAGE_URL="http://solo-packages.s3-website-us-east-1.amazonaws.com"

This repository was created on 9/3/2015 and was built for kernel:
2015-09-03T21:06:15.000Z        3.5 MB         kernel-image-3.10.17-rt12-1.0.0-ga+g3f15a11-3.10.17-r0.imx6solo_3dr_1080p.rpm

Assuming that you have installed the v4l2loopback module using smart,
when you try to execute:
"modprobe v4l2loopback exclusive_caps=0,0 video_nr=1,2"

You will get:
"modprobe: FATAL: Module v4l2loopback not found."

That is because smart installed the module from the repository that was NOT compiled for the kernel that you are running.
That is because 3dr has not updated the repository to correspond to the kernel that is being distributed in solo firmware 1.3.0.

This worked before the 1.3.0 update because the kernel and module revisions matched.

3dr needs to update the repositories.
3dr also needs to provide the source for the v4l2loopback module, as it is not present in:
"Open Source Components: solo-open-source-components-1.1.12.tar.gz , solo-open-source-components.tar.gz"

This code is available from:
https://github.com/umlaeute/v4l2loopback

franknitty69, 2016-01-16:
great catch.

hamishwillee, 2016-01-28:
Hi @GPSFan 

Thanks for bringing this to our attention. We don't yet have toolchain in place to allow regular rebuilds and keep the binaries updated. 

We'll get an engineer to look at what needs to be done for this (and to look at issues around the module source) in the coming month. I can't provide an ETA on an actual fix until that has been done. 

@mrpollo - job for you to assign that engineer :-)

GPSFan, 2016-01-28:
I'd be more than willing to test what you come up with, I've been doing linux kernel work for the last 12 years or so, and have used lots of different toolchains, glibs & build systems. I do shy away from proprietary stuff, and prefer FOSS, hence my attraction to Solo...

hamishwillee, 2016-01-29:
Thanks @GPSFan 

@tcr3dr is assigned to tidying up the docs and working out a way to progress this (and a number of related issues).

hamishwillee, 2016-02-11:
@GPSFan We're going to do some work to support this use case properly with public APIs. In the meantime we've decided to "unpublish" the current solution (#281). I am therefore closing this issue.

closed 2016-02-11 by hamishwillee

GPSFan, 2016-02-12:
Thanks,
Hope to see some new API's in the near future...

hamishwillee, 2016-02-12:
@GPSFan Me too Ken. I'm sorry I can't give you a real ETA (though I think this will be at least a few months). At least when we do deliver this it will be something you can rely on.