3drobotics/solodevguide #258: Smart Battery SMBUS Communication

franknitty69, 2016-01-19:
@mrpollo i have been using any and all methods of smbus hardware trying to talk to the smart battery for the last 3 months. no matter what interface i use, i cannot establish communication with the smart battery.

i've tried an arduino uno r3 with every i2c library available. i have tried the texas instruments EV2300 and 2400 with the evaluation software on both i2c and smbus ports. they are connected to the solo battery via Molex 171088-0048 and yes i have verified that data and clock lines are connected correctly.

simply connecting the battery and pressing the power button should register an acknowledgment on the bus but its never received. and any read or write words result in no ack.

can you provide some insight?

thanks

mrpollo, 2016-01-22:
hey @franknitty69 we added new info regarding the battery bay http://dev.3dr.com/hardware-batterybay.html, let me know if this helps

franknitty69, 2016-01-26:
the ev2300 works now! i found out from texas instrument tech support that the ground wire for the smb connection has to be connected to the smart battery ground.

hamishwillee, 2016-01-26:
@franknitty69 Thanks for letting us know! I'd like to make sure this information is available on  http://dev.3dr.com/hardware-batterybay.html and/or http://dev.3dr.com/hardware-battery-charging.html and I was wondering if you could give me some advice on how to update the docs?

My thought was that what we currently specify is just the power connections - we don't mention the SMBUS connector - is that how you see it? I think what we need to do is add information about the [smbus connector here](http://dev.3dr.com/hardware-batterybay.html#electrical-interface-specifications) and then add your information as a note at the end of the section? Does that make sense?

hamishwillee, 2016-01-26:
PS other than that, was the information provided in our battery bay and battery charger documentation sufficient for you to work with the battery? Is there any other information that you thought was missing, or which you thought could be added to improve our offering in this area?

franknitty69, 2016-01-27:
sure thing. the connector is already specified on http://dev.3dr.com/hardware-battery-charging.html.

What needs to be added to the documentation is which blade is sda and scl. also the fact that the battery and interface must share a common ground. i think anything more detailed than that will be out of the scope of the dev guide.

steveheinsch, 2016-02-02:
@franknitty69 So, if the battery is sitting on the table with the power button side facing up, which side of the data port is SDA and which is SCL? Top/Bottom? Thanks

franknitty69, 2016-02-02:
the solo battery has female connection blades smbus, which is split blade (2 circuits) and the other two are regular blades (positive and negative). for the smbus connector, with the power button facing up, the top blade is clock (scl) and the bottom blade is data (sda).

hamishwillee, 2016-02-02:
@franknitty69 Thanks. I don't really understand what the split blade/2 circuit thing means. I THINK that you're saying that we have three blades, 2 are used for power (positive and negative) and 2 are used by SMBUS. Correct?

Just to confirm, is the diagram below correct?
[image]

steveheinsch, 2016-02-03:
@hamishwillee No that diagram isn't correct.

Turn the battery so it sits with the Power Button on top (opposite of your picture). There are 3 connectors, or "ports" as you referred to them.

Top Connector = SMBus connector, which is divided into 2 sections internally - the top and the bottom.  The top 3 contacts of that (single) port are connected to SCL/Clock. The bottom 3 contacts are SDA/Data. That receptacle is also a bit larger then the 2 battery power connectors.

Middle Connector = Battery Positive (there's a + next to the port)
Bottom Connector = Battery Ground (there's a - next to the port)

GPSFan, 2016-02-03:
Like this:
[image]

steveheinsch, 2016-02-03:
@GPSFan Yes, sir!

hamishwillee, 2016-02-03:
Awesome. Thanks all. And the third blade is ground?

steveheinsch, 2016-02-03:
Yes, bottom blade is negative/ground

franknitty69, 2016-02-03:
when i was referring to blades i was talking from the male connector side. sorry for the confusion.

hamishwillee, 2016-02-03:
Hi @franknitty69 et al
No, pretty much my fault for not understanding what was meant by blade. That's the joy of coming from a software background. Thanks everyone for the explanation!

I've updated the image with the scl/sda labels and some text. This is on both the charger and bay pages. You can see it on #275 (please review!) or rendered [here](https://dl.dropboxusercontent.com/u/3067678/share2/sdg-batsmbus/_book/hardware-battery-charging.html#communication-protocol).

franknitty69, 2016-02-03:
the image looks good

hamishwillee, 2016-02-10:
This is now fixed and live

closed 2016-02-10 by hamishwillee