3drobotics/solodevguide #263: travis reports wrong links in /advanced-logs.html

mrpollo, 2016-01-26:
we need to find why travis thinks this is wrong
```
[Tue, 26 Jan 2016 08:16:05 GMT] "GET /advanced-logs.html#downloading-using-scp-mac-linux" "Node/simplecrawler 0.4.2 (git+ssh://git@github.com/cgiffard/node-simplecrawler.git)"
>> Error finding content with the following fragment identifier linked from http://localhost:4000/advanced-logs.html to http://localhost:4000/advanced-logs.html#downloading-using-scp-mac-linux
```

mrpollo, 2016-01-26:
Just an example of a failed build https://travis-ci.org/3drobotics/solodevguide/builds/104838901

hamishwillee, 2016-01-26:
@mrpollo So as @OXINARF indicates in #264 the problem is that the anchor is generated differently locally and on production:
#downloading-using-scp-maclinux (Development) 
#downloading-using-scp-mac-linux (Production) 

Upshot, this is not a test problem, but something to do with the different build environment locally and on production.

mrpollo, 2016-01-26:
I'm not sure what is causing this problems but this is absolute nuts, it's driving me mad!, I'll take a deeper look today and try to get to the bottom of this

hamishwillee, 2016-01-26:
:-) I'd start with comparing the build environments for local and server builds. Now I think about it ... that's what you were asking for the other day. Good luck, this sort of thing is a huge waste of time!

PR #264 (2016-02-01): Wrong link, try to fix tests
Fixes #263
commit 1d2413f1 (2016-02-01): Wrong link, try to fix tests

Fixes #263
closed 2016-02-01 by hamishwillee