3drobotics/solodevguide #271: USB to Gimble connection

GPSFan, 2016-02-01:
I'm not sure this is the right place to raise this issue as it is really hardware related, but since the architectural overview was just added, I'll start here.
The overview shows a USB connection from the SoloLink CPU to the Accessory bay, as well as one to the Gimbal Bay.
You gotta open up the Solo to see the following.
The USB to the Accessory bay is a  shielded twisted pair as is required by the USB spec.
The USB to the Gimbal Bay is neither shielded, nor twisted. It is also bundled with all the other signal and power wires going to the Gimbal.  Even though the cable bundle is short (~5") the USB High Speed signal fidelity can be severely reduced by even such a short run, especially with other signals in the bundle. If the USB to the Gimbal Bay is being used to just power/charge the camera that's one thing, but as there are D+ and D- wires in that bundle, actually using the USB as USB might show some data integrity problems.
I know that getting all the signals to the Gimbal Bay had many challenges, but developers need to be aware of the potential of USB problems with the Gimbal.

hamishwillee, 2016-02-04:
Thanks @GPSFan . You've right. I've added a note in #277 to highlight the potential impact.

GPSFan, 2016-02-04:
Great, hopefully, future rev's of the Solo main board and wiring harness will improve this.
You can close this issue.
Thanks...

closed 2016-02-04 by hamishwillee