3drobotics/solodevguide #282: MAV ID change

mangritosh, 2016-02-15:
Hi, 

I've changed SYSID_THISMAV using Mission Planner for SOLO, but the controller stopped connecting to the drone (says 'waiting for solo'). Is there a way to change SYSID_THISMAV parameter so the controller would still recognise the drone? 

Thanks!

hamishwillee, 2016-02-15:
Almost certainly not. On quick inspection the id for messages sent the the vehicle appears to be hard coded.

Why do you want to change the ID?

mangritosh, 2016-02-15:
I have two vehicles that I'd like to connect to APM planner and that would have different MAV id's. Is there any other way?

hamishwillee, 2016-02-16:
Solo is pretty much a Linux box with wings and the ground control software is opensource. So in theory you can do whatever you want, but it might require a re-write of major parts of the system. This is certainly not something 3DR would support you with.

I don't know what all these dependencies are. The most obvious I can think of is dronekit - Solo uses a very old version of DroneKit (for smartshots) that had a fixed target ID. 

> Why would this be hard-coded? 

Given there tight relationship between the controller and Solo on their own Wifi network and Solo's product design goals I suspect that there just wasn't any thinking that support for different IDs might be needed.

umarj, 2016-02-16:
Hey Hamish, 

Are the IPs hardcoded too? Is it possible to change the IPs of both the controller and the drone itself and still have them communicating? If the IPs can be changed, will it require hacking around Yokto to get them to communicate again? Can you please point us in the right direction here. 

There is a video of swarming Solos which Chris Anderson put up on DIY drones. Surely, they must have changed MAV IDs and IPs to do this.
http://diydrones.com/profiles/blogs/swarming-solos-over-berkeley?xg_source=activity

We bought a few Solos to do something similar and it is painful to find out that these things are hard coded. Any help would he highly appreciated.

Thanks.

hamishwillee, 2016-02-16:
Hi @umarj 

Probably. Here are a few words on that swarming solos implementation from the team that created it (as an experiment):

> To make it work we killed pretty much everything that interacted with the network. Then we used a custom version of Telem forwarder that sent raw mavlink packets, and hardcoded the ip address and destination ports for each Solo.

> After that you can open mavproxy or dronekit sessions to each drone by adjusting the ip and destination port.

This is one of the things that would actually be a lot easier to do with a less productised platform like the 3DR Iris and using the latest dronekit.

As above we really can't "support" this (I won't be able to get more information on this). That said, I hope this helps, and I'd be very pleased to see you being successful :-)

Closing this now.

closed 2016-02-16 by hamishwillee