3drobotics/solodevguide #283: Cannot install opencv requirement

mpetkova, 2016-02-17:
When running solo script pack for the opencv examples on OS X 10.8.5:

`Downloading/unpacking opencv==2.4.8 (from -r ./requirements.txt (line 1))
  Could not find any downloads that satisfy the requirement opencv==2.4.8 (from -r ./requirements.txt (line 1))
Cleaning up...
No distributions at all found for opencv==2.4.8 (from -r ./requirements.txt (line 1))`

mpetkova, 2016-02-18:
Getting the same error when using Ubuntu 15.04

hamishwillee, 2016-02-19:
Have you already run `smart install python-opencv` on the vehicle?

I get an error for missing numpy when it tries to run, but otherwise it works.

Are you using the latest version of solo-cli and solo?

mpetkova, 2016-02-19:
I did install python-opencv on the vehicle and just updated solo-cli and solo a couple of days ago. Packaging only worked when I replaced that requirement with one of [these pyopencv ones](https://pypi.python.org/pypi/pyopencv). Then I also got an error for numpy.

I'm generally having trouble with solo script pack on OS X. The only way I could run the opencv examples successfully was onboard the drone.

hamishwillee, 2016-02-19:
We've just improved solo script pack on all the OS, and it works for at least helloworld. I suspect the problem in this case is that there are no binaries that can be used for numpy. In any case, I've asked my more-expert colleague (@peter.barker) to have a look at this in the coming week.

mpetkova, 2016-02-19:
I may have gotten confused with all the examples I tried to run, but you're right, helloworld.py does work for me using solo script run. On board the drone, I keep getting the error:
`root@3dr_solo:~/helloworld# python helloworld.py
Traceback (most recent call last):
  File "helloworld.py", line 1, in <module>
    from dronekit import connect
ImportError: No module named dronekit`

It looks like I'll be needing dronekit extensively - any idea what the problem is?

SoloCamera.py on the other hand only works when run on board. 

I'm not sure if this is the right place to ask, but I'm working on a project that requires opencv and I'm having a lot of trouble with that. I can see the video stream through the mobile app, however, on board Solo:

`root@3dr_solo:~# lsusb
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub`

`>>> import cv2
>>> 
>>> cam = cv2.VideoCapture(0)
>>> print cam.isOpened()
False
>>> `

It seems like I can't access the camera command-line at all. Do you have any idea what the issue might be or where I could look for help on this? Thank you.

hamishwillee, 2016-02-21:
@mpetkova When did you last update solo-cli? Helloworld does not give me that error as of yesterday. Can you please update solo-cli from github and test that again?

As to OpenCV, this is the right place to ask about examples, but I am not the right person to sort this issue out. I'll ask others in the team to help, but we may not be fast.

mpetkova, 2016-02-21:
I'm running the most up-to-date version I believe. This is what I just ran:

`sudo -H pip install -U git+https://github.com/3drobotics/solo-cli
Password:
Collecting git+https://github.com/3drobotics/solo-cli
  Cloning https://github.com/3drobotics/solo-cli to /tmp/pip-kRFHWN-build
Requirement already up-to-date: docopt>=0.6.2 in /usr/local/lib/python2.7/site-packages (from solo-cli==1.1.2)
Requirement already up-to-date: paramiko>=1.15.2 in /usr/local/lib/python2.7/site-packages (from solo-cli==1.1.2)
Requirement already up-to-date: scp>=0.10.0 in /usr/local/lib/python2.7/site-packages (from solo-cli==1.1.2)
Requirement already up-to-date: requests>=2.5.0 in /usr/local/lib/python2.7/site-packages (from solo-cli==1.1.2)
Requirement already up-to-date: pyopenssl in /usr/local/lib/python2.7/site-packages (from solo-cli==1.1.2)
Requirement already up-to-date: ndg-httpsclient in /usr/local/lib/python2.7/site-packages (from solo-cli==1.1.2)
Requirement already up-to-date: pyasn1 in /usr/local/lib/python2.7/site-packages (from solo-cli==1.1.2)
Requirement already up-to-date: ecdsa>=0.11 in /usr/local/lib/python2.7/site-packages (from paramiko>=1.15.2->solo-cli==1.1.2)
Requirement already up-to-date: pycrypto!=2.4,>=2.1 in /usr/local/lib/python2.7/site-packages (from paramiko>=1.15.2->solo-cli==1.1.2)
Requirement already up-to-date: six>=1.5.2 in /usr/local/lib/python2.7/site-packages (from pyopenssl->solo-cli==1.1.2)
Requirement already up-to-date: cryptography>=0.7 in /usr/local/lib/python2.7/site-packages (from pyopenssl->solo-cli==1.1.2)
Requirement already up-to-date: enum34 in /usr/local/lib/python2.7/site-packages (from cryptography>=0.7->pyopenssl->solo-cli==1.1.2)
Requirement already up-to-date: ipaddress in /usr/local/lib/python2.7/site-packages (from cryptography>=0.7->pyopenssl->solo-cli==1.1.2)
Requirement already up-to-date: setuptools>=1.0 in /usr/local/lib/python2.7/site-packages (from cryptography>=0.7->pyopenssl->solo-cli==1.1.2)
Requirement already up-to-date: idna>=2.0 in /usr/local/lib/python2.7/site-packages (from cryptography>=0.7->pyopenssl->solo-cli==1.1.2)
Requirement already up-to-date: cffi>=1.4.1 in /usr/local/lib/python2.7/site-packages (from cryptography>=0.7->pyopenssl->solo-cli==1.1.2)
Requirement already up-to-date: pycparser in /usr/local/lib/python2.7/site-packages (from cffi>=1.4.1->cryptography>=0.7->pyopenssl->solo-cli==1.1.2)
Installing collected packages: solo-cli
  Found existing installation: solo-cli 1.1.2
    Uninstalling solo-cli-1.1.2:
      Successfully uninstalled solo-cli-1.1.2
  Running setup.py install for solo-cli ... done
Successfully installed solo-cli-1.1.2`

When running helloworld.py on my computer:

`python helloworld.py 
Connecting to Solo ...
>>> APM:Copter solo-1.2.21 (aec18c35)
>>> PX4: 5e693274 NuttX: d48fa307
>>> Frame: QUAD
>>> PX4v2 00370016 3334510F 32363238
>>> PreArm: Need 3D Fix
 Mode: LOITER
Global Location: LocationGlobal:lat=42.3597904,lon=-71.0937888,alt=0.41
Global Location (relative altitude): LocationGlobalRelative:lat=42.3597904,lon=-71.0937888,alt=0.41
Local Location: LocationLocal:north=None,east=None,down=None
Attitude: Attitude:pitch=-0.00478100916371,yaw=-0.10869833827,roll=0.0190431065857
Velocity: [0.0, 0.0, 0.03]
GPS: GPSInfo:fix=3,num_sat=8
Groundspeed: 0.070000000298
Airspeed: 0.070000000298
Battery: Battery:voltage=15.392,current=0.8,level=63
EKF OK?: False
Last Heartbeat: 0.3120470047
Rangefinder: Rangefinder: distance=None, voltage=None
Rangefinder distance: None
Rangefinder voltage: None
Heading: 353
Is Armable?: False
System status: STANDBY
Mode: LOITER
Armed: False
Traceback (most recent call last):
  File "helloworld.py", line 43, in <module>
    time.sleep(10)
NameError: name 'time' is not defined
`

When running helloworld.py using solo script pack / solo script run:

`solo script run helloworld.py 
connecting to solo...
checking pip...
[... 29 lines cut ...]
Exception in thread Thread-2 (most likely raised during interpreter shutdown):
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 551, in __bootstrap_inner
  File "/usr/lib/python2.7/threading.py", line 504, in run
  File "/log/solo-script/env/lib/python2.7/site-packages/dronekit/mavlink.py", line 89, in mavlink_thread_out
<type 'exceptions.AttributeError'>: 'NoneType' object has no attribute 'error'
Exception in thread Thread-1 (most likely raised during interpreter shutdown):
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 551, in __bootstrap_inner
  File "/usr/lib/python2.7/threading.py", line 504, in run
  File "/log/solo-script/env/lib/python2.7/site-packages/dronekit/mavlink.py", line 124, in mavlink_thread_in
  File "/log/solo-script/env/lib/python2.7/site-packages/dronekit/__init__.py", line 1295, in listener
<type 'exceptions.AttributeError'>: 'NoneType' object has no attribute 'time'`

And finally, when running helloworld.py onboard the drone:

`root@3dr_solo:~/helloworld# python helloworld.py
Traceback (most recent call last):
  File "helloworld.py", line 1, in <module>
    from dronekit import connect
ImportError: No module named dronekit`

I know my original question was about installing the opencv requirement, so if you think I should open up new issues about this and the issue of the drone not recognizing the GoPro I can do so. 

Thanks again.

DarylWM, 2016-03-23:
Just getting back to why `solo script pack` doesn't work for the opencv example, is there a fix for the "No matching distribution found for opencv==2.4.8" issue?

tasarmiento, 2016-10-13:
Did anyone determine a solution for this issue ("No matching distribution found for opencv==2.4.8")?