3drobotics/solodevguide #284: Gopro functions for Dronekit?

nicOwlas, 2016-02-18:
Hi,

I am not sure it is the right place to raise the question but I did not find any better:

I would like to control my Gopro onboard the solo via some dronekit functions and I am therefore looking for the functions to do this. This would allow me to trigger the camera at specific GPS positions etc.

I have found this blog from Daniel McKinnon: http://www.ddmckinnon.com/2015/12/30/idiots-guide-to-dronekit-python-a-journey-to-whoz-chillin/ which gives mavlink messages syntax to change Gopro mode, take picture and control the Gimbal.

**Suggested function to take a picture:**
`    def goproTakePic(self):`
`        MAVLINK_GIMBAL_SYSTEM_ID=1`
`        MAVLINK_GIMBAL_COMPONENT_ID=154`
``        
`        msg = self.message_factory.gopro_set_request_encode(`
`        MAVLINK_GIMBAL_SYSTEM_ID,`
`        MAVLINK_GIMBAL_COMPONENT_ID,`
`        mavutil.mavlink.GOPRO_COMMAND_SHUTTER,`
`        (1, 0, 0, 0))`
``
`        self.send_mavlink(msg)`
`        self.flush()`

The Gimbal control works (sad though that the default dronekit gimbal functions do not work with solo). The Gopro functions doesn't seem to work for me (*cygwin*, *ubuntu*, *solo-cli 1.1.2*). Is there any more documentation somewhere on how to control the Gopro?

Digging into solo code, I found in `./usr/bin/` `shotManager.py` and `GoProManager.py` which seems to show that it is feasible but any way to access those functions from dronekit?

Thanks! :-)

hamishwillee, 2016-02-19:
II have no idea. I would perhaps try using MAVProxy as it appears to support a gopro module (https://github.com/Dronecode/MAVProxy/blob/master/MAVProxy/modules/mavproxy_gopro.py). If it works, then you can engineer what commands work and what do not.

hamishwillee, 2016-02-19:
Looking around at the code, the trick is probably to send messages using the correct cmponent ID.
``mavutil.mavlink.MAV_COMP_ID_GIMBAL``.

E.g. to send a request ...
```

        msg = self.message_factory.gopro_get_request_encode(
                                    0, mavutil.mavlink.MAV_COMP_ID_GIMBAL,    # target system, target component
                                    command
                                    )
        self.send_mavlink(msg)
```

hamishwillee, 2016-02-19:
Enough to get started?

nicOwlas, 2016-02-19:
Hi @hamishwillee!

Yes thanks a lot, just got into office. I'll have a look today itself and let you know!

Cheers,
nicOwlas

nicOwlas, 2016-02-19:
Hi @hamishwillee ,

Short status: nothing worked and after digging everything up, it seems that the problem comes from my gimbal (I cannot even control the gopro from the official app). 3DR aftersale said that it was damaged, so I returned the gimbal and I will wait for the new one to give it another try. Hope next week.

I'll let you know,

Cheers

hamishwillee, 2016-02-21:
Thanks for letting me know @nicOwlas. Sorry to hear about the gimbal!

imutkarshpatil, 2017-10-26:
Hey @nicOwlas ,

Were you able to access GoPro after the setting new gimbal?

nicOwlas, 2017-10-26:
Hi @imutkarshpatil,

Ahah, I was not expecting a message on this issue after 18 months ! To be fully honest no, but we did not investigate for too long, we switched to DJI.