3drobotics/solodevguide #286: Running helloworld.py on Solo gives exception, but runs fine locally

DarylWM, 2016-02-22:
Wondering if anybody has come across this exception when running helloworld.py on Solo? The script finishes, but raises the exception on shutdown. The script runs fine locally.

...
 Mode: LOITER
 Is Armable?: False
 Armed: False
Done.
Exception in thread Thread-2 (most likely raised during interpreter shutdown):
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 551, in __bootstrap_inner
  File "/usr/lib/python2.7/threading.py", line 504, in run
  File "/log/solo-script/env/lib/python2.7/site-packages/dronekit/mavlink.py", line 89, in mavlink_thread_out
<type 'exceptions.AttributeError'>: 'NoneType' object has no attribute 'error'
Exception in thread Thread-1 (most likely raised during interpreter shutdown):
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 551, in __bootstrap_inner
  File "/usr/lib/python2.7/threading.py", line 504, in run
  File "/log/solo-script/env/lib/python2.7/site-packages/dronekit/mavlink.py", line 129, in mavlink_thread_in
<type 'exceptions.AttributeError'>: 'NoneType' object has no attribute 'error'

hamishwillee, 2016-02-22:
Thanks @DarylWM Yes, this is a known issue.

pjbca, 2016-04-04:
Does this prevent connection on running the script again?
I am getting sporadic connection between the python script running on solo, and the vehicle.
A typical error is:
  File "/home/root/opt/my_python_code/env/lib/python2.7/site-packages/dronekit/__init__.py", line 2142, in wait_ready
    timeout)
dronekit.APIException: wait_ready experienced a timeout after 30 seconds.

keyurrakholiya, 2016-06-11:
i am facing same problem.suggest help

janthony, 2016-07-07:
I am struggling with the same issue. Any proposed workaround for the time being?

tasarmiento, 2017-03-02:
Did anyone find a solution to this issue?  I'm seeing it with my own scripts running onboard Solo.