3drobotics/solodevguide #288: Disabling of host key checking does not appear to work

hamishwillee, 2016-02-22:
At least on Mac/Linux you need to have connected to solo via SSH at least once for solo-cli to connect.

@tcr3dr indicates that this may be [related to known host key checking](http://linuxcommando.blogspot.com/2008/10/how-to-disable-ssh-host-key-checking.html) issues. This should be corrected by this line of code but we need to confirm https://github.com/3drobotics/solo-cli/blob/master/soloutils/__init__.py#L27

kevinmehall, 2016-02-22:
The `missing_host_key_policy` is invoked only when the host key is not found in known_hosts. If it is found and doesn't match, it is rejected before that function is called.

hamishwillee, 2016-02-22:
@kevinmehall So what are you saying? In this case I think that the client (PC) doesn't have the key because it has never connected to the host (Solo) so the missing_host_key_policy as set *should* just auto-add the key. What appears to be happening is that we're not connecting because the key is not being auto-added.

kevinmehall, 2016-02-22:
The first time you connect to 10.1.1.10, the host key for that IP is not found, and it invokes the auto_add policy, which adds it to known_hosts. Then you connect to a different Solo (which is also 10.1.1.10 because it's a different network), and it finds the first Solo's key in known_hosts and rejects the connection before invoking the policy. What I'm saying is that the `missing_host_key_policy` is not the appropriate mechanism to do this because it never gets called in the failure case.

kevinmehall, 2016-02-22:
Try removing the call to `client.load_system_host_keys()`

hamishwillee, 2016-02-22:
@kevinmehall This does look like a problem - sounds like load_host_keys() would be better than the load_system_host_keys().

I don't think this solves the problem reported above though - that the user has to log into solo at least once before they can use solo cli - and in this case they have never logged in to another solo.
Correct?

Ie need to have a separate defect for your comment?

kevinmehall, 2016-02-23:
Ah yes, I read the title and assumed it was the problem I've experienced. Auto-add or a no-op function should accept the key for an IP that has not been seen before, so this doesn't explain the issue described in the initial comment.