3drobotics/solodevguide #291: please help me//  synchronizing code and files between your desktop and Solo

orhong, 2016-02-24:
[ http://dev.3dr.com/starting-installing.html]
I had to run: rsync -avz local/file/path/. root@10.1.1.10:/solo/path/.
but this occur error.....

pi@raspberrypi:~ $  rsync -avz local/file/path/. root@10.1.1.10:/solo/path/.
root@10.1.1.10's password: TjSDBkAu
sending incremental file list
rsync: change_dir "/home/pi//local/file/path" failed: No such file or directory (2)
rsync: change_dir#3 "/solo/path" failed: No such file or directory (2)
rsync error: errors selecting input/output files, dirs (code 3) at main.c(625) [Receiver=3.0.9]
rsync: connection unexpectedly closed (175 bytes received so far) [sender]
rsync error: error in rsync protocol data stream (code 12) at io.c(226) [Sender=3.1.1]

... what's the problem 
???

franknitty69, 2016-02-24:
the problem is that 'local/file/path doesn't exist on your rasberry pi. nor does it exist on your solo. 

the code isn't literal, it just to show you how it is done. @hamishwillee i guess this should be made more clearer.

**example**
rasberry pi /orhong/mycode/mycode.py
solo /mycode

rsync -avz /orhong/mycode/. root@10.1.1.10:/mycode
this will copy mycode.py from /orhong/mycode to the solo at /mycode. if mycode doesen't exist it will be created.

also rsync is the old method, the new recommended method of moving code to solo for purposes of executing it is solo script pack and solo script run.

hamishwillee, 2016-02-24:
@orhong As @franknitty69 says, the **local/file/path/** and **/solo/path/* are a very common shorthand for "the path where your you want to copy is located" and "the location on  where you want to copy the file"

Would the following have been more clear for you (and you @franknitty69)?

```
rsync -avz source-file-path-on-local-computer root@10.1.1.10:/destination-file-path-on-solo
```

If you're working with python then the best tool to use is solo cli - see http://dev.3dr.com/starting-installing.html#installing-packages-using-the-solo-client

I think I'll reorganise the page to make that clear as python is probably the main use case.