3drobotics/solodevguide #292: Enabling serial 2 for telemetry breaks gimbal

sondree, 2016-02-25:
Hi!

We are developing a payload for the 3DR Solo that requires telemetry access to the Pixhawk. By enabling SER2 (on the accessory port), first by creating _/fs/microsd/APM/uartD.en_ and then setting parameter _SERIAL2_PROTOCOL_ to 1. The gimbal pitch control stops working. 

Is the SER2 port on the accessory bay somehow connected to control of the official gimbal?

Hope to hear from you soon. 

In advance thanks

NiharikaPatel, 2019-08-20:
@sondree did you ever find a way around this? I have the same issue.

Thanks

sondree, 2019-08-21:
No, nothing short of dkwngrading2 firmware unfortunately

NiharikaPatel, 2019-08-22:
Hi @sondree It seems like my previous comment got deleted
 "@sondree  did you ever find a way around this? I have the same issue.
 Thanks"
But thank you for you response. I  currently have Open Solo. Which firmware did you downgrade to for it to work?

sondree, 2019-08-22:
On the official firmware 2.3.0-1 is the last before it breaks integration.
On the top of my head I dont remember which ardupilot firmware that
corresponds to.

Are you saying that even with the open ardupilot fork serial is broken for
later releases?

Please be aware that Serial2 has to be enabled both through serial in the
flight controller shell and through a mavlink param to work.

sondree, 2019-08-22:
To clarify, firmwares after 2.3.0-1 breaks Serial2 altogether. I have not
been able to use the gimbal on any firmware releases. Related issue being
that later firmwares disable access to Serial2

NiharikaPatel, 2019-08-27:
Hi @sondree we updated to Open Solo and then  AC3.7 dev and everything seems to work fine now. We are able to use Serial2 to connect it to our Rpi and the gimbal still works even after changing the SERIAL2_PROTOCOL to 1

sondree, 2019-08-27:
Nice to know! Thanks for the update