3drobotics/solodevguide #297: RequirementParseError: Expected version spec in ./wheelhouse/* at /wheelhouse/*

alexblack, 2016-03-22:
Hi, I'm trying to follow along here:

https://dev.3dr.com/advanced-python.html

Logs from solo showing error:

```
root@3dr_solo:/opt# cd my_python_code/
root@3dr_solo:/opt/my_python_code# virtualenv env
New python executable in env/bin/python
Installing setuptools, pip, wheel...done.
(env)root@3dr_solo:/opt/my_python_code# source ./env/bin/activate
(env)root@3dr_solo:/opt/my_python_code# pip install --no-index ./wheelhouse/* -UI
Ignoring indexes: https://pypi.python.org/simple
Exception:
Traceback (most recent call last):
  File "/opt/my_python_code/env/lib/python2.7/site-packages/pip/basecommand.py", line 211, in main
    status = self.run(options, args)
  File "/opt/my_python_code/env/lib/python2.7/site-packages/pip/commands/install.py", line 282, in run
    wheel_cache
  File "/opt/my_python_code/env/lib/python2.7/site-packages/pip/basecommand.py", line 272, in populate_requirement_set
    wheel_cache=wheel_cache
  File "/opt/my_python_code/env/lib/python2.7/site-packages/pip/req/req_install.py", line 213, in from_line
    wheel_cache=wheel_cache, constraint=constraint)
  File "/opt/my_python_code/env/lib/python2.7/site-packages/pip/req/req_install.py", line 67, in __init__
    req = pkg_resources.Requirement.parse(req)
  File "/opt/my_python_code/env/lib/python2.7/site-packages/pip/_vendor/pkg_resources/__init__.py", line 2980, in parse
    reqs = list(parse_requirements(s))
  File "/opt/my_python_code/env/lib/python2.7/site-packages/pip/_vendor/pkg_resources/__init__.py", line 2924, in parse_requirements
    "version spec")
  File "/opt/my_python_code/env/lib/python2.7/site-packages/pip/_vendor/pkg_resources/__init__.py", line 2889, in scan_list
    raise RequirementParseError(msg, line, "at", line[p:])
RequirementParseError: Expected version spec in ./wheelhouse/* at /wheelhouse/*
```

Here are some logs from my laptop:

```
(env) Alexs-MacBook-Air:solo alex$ pip freeze > requirements.txt
(env) Alexs-MacBook-Air:solo alex$ pip wheel -r ./requirements.txt --build-option="--plat-name=py27"
/Users/alex/dev/ddc/solo/env/lib/python2.7/site-packages/pip/commands/wheel.py:126: UserWarning: Disabling all use of wheels due to the use of --build-options / --global-options / --install-options.
  cmdoptions.check_install_build_global(options)
Collecting dronekit==2.4.0 (from -r ./requirements.txt (line 1))
  Using cached dronekit-2.4.0.tar.gz
Collecting monotonic==0.6 (from -r ./requirements.txt (line 2))
  Using cached monotonic-0.6.tar.gz
Collecting pymavlink==1.1.72 (from -r ./requirements.txt (line 3))
  Using cached pymavlink-1.1.72.tar.gz
Building wheels for collected packages: dronekit, monotonic, pymavlink
  Running setup.py bdist_wheel for dronekit ... done
  Stored in directory: /Users/alex/dev/ddc/solo
  Running setup.py bdist_wheel for monotonic ... done
  Stored in directory: /Users/alex/dev/ddc/solo
  Running setup.py bdist_wheel for pymavlink ... done
  Stored in directory: /Users/alex/dev/ddc/solo
Successfully built dronekit monotonic pymavlink
(env) Alexs-MacBook-Air:solo alex$ rsync -avz --exclude="*.pyc" --exclude="env" ./ root@10.1.1.10:/opt/my_python_code
root@10.1.1.10's password:
building file list ... done
created directory /opt/my_python_code
./
dronekit-2.4.0-py2-none-py27.whl
hello_solo.py
monotonic-0.6-py2-none-py27.whl
pymavlink-1.1.72-py2-none-py27.whl
requirements.txt

sent 1914089 bytes  received 136 bytes  294496.15 bytes/sec
total size is 1937897  speedup is 1.01
```

alexblack, 2016-03-22:
using `pip install --no-index *.whl -U` seems to work since there is no `.wheelhouse` folder (that seems like a solo-cli thing) but, then I get #298

OXINARF, 2016-03-23:
According to the guide, those first commands should be run in your laptop and not on Solo.

alexblack, 2016-03-23:
Read the link I provided :)  You end up running similar instructions on both.

OXINARF, 2016-03-23:
Ups, sorry, the page hadn't completely loaded so I missed it. May I ask why you are trying to do this instead of using solo-cli pack?

alexblack, 2016-03-23:
Tried that first, hit https://github.com/dronekit/dronekit-python/issues/524

OXINARF, 2016-03-23:
It seems the issue is that it is packaging the wrong pymavlink. Can you post the log from running `solo script pack`?

alexblack, 2016-03-23:
```
(ddc) Alexs-MacBook-Air:solo alex$ solo script pack
checking Internet connectivity...
Creating script archive...
site packages directory: /var/folders/df/wzvlvcy95hjbvjmwljpf8q280000gn/T/tmpwaYJ5w/content/env/lib/python2.7/site-packages
script is at /var/folders/df/wzvlvcy95hjbvjmwljpf8q280000gn/T/tmpoL5mJg
Requirement already satisfied (use --upgrade to upgrade): wheel in ./env/lib/python2.7/site-packages
script is at /var/folders/df/wzvlvcy95hjbvjmwljpf8q280000gn/T/tmpLpsZSz
Collecting dronekit==2.4.0 (from -r requirements.txt (line 1))
  Saved ./wheelhouse/dronekit-2.4.0-cp27-none-any.whl
Collecting pymavlink>=1.1.62 (from dronekit==2.4.0->-r requirements.txt (line 1))
  Saved ./wheelhouse/pymavlink-1.1.72-cp27-cp27m-macosx_10_11_x86_64.whl
Collecting monotonic<1.0 (from dronekit==2.4.0->-r requirements.txt (line 1))
  Saved ./wheelhouse/monotonic-0.6-cp27-none-any.whl
Skipping dronekit, due to already being wheel.
Skipping pymavlink, due to already being wheel.
Skipping monotonic, due to already being wheel.
Creating tarball (solo-script.tar.gz)
```

OXINARF, 2016-03-25:
Can you run this command in Solo: `python -c "import distutils.util; print(distutils.util.get_platform())"`?

alexblack, 2016-03-30:
root@3dr_solo:/opt/ddc# python -c "import distutils.util;
print(distutils.util.get_platform())"
linux-armv7l

OXINARF, 2016-03-30:
Ok, when you execute the line `pip wheel -r ./requirements.txt --build-option="--plat-name=py27"` can you replace the _py27_ with the _linux-armv7l_?