3drobotics/solodevguide #310: VISION_POSITION_ESTIMATE

alexblack, 2016-05-18:
Does the Solo support the VISION_POSITION_ESTIMATE message? See: http://dev.px4.io/external-position.html

I'm thinking (hoping?) that it does, since the Solo uses PX4 firmware?

Thanks!

OXINARF, 2016-05-18:
Solo uses ArduPilot, not PX4.

alexblack, 2016-05-18:
Huh, I obviously still have a lot to learn.

1. The question still remains, does the Solo implement the VISION_POSITION_ESTIMATE message?

2. Follow on question - from what I've seen Solo uses PX4 *and* ArduPilot?  eg https://github.com/3drobotics/PX4Firmware-solo

OXINARF, 2016-05-19:
1. No, ArduPilot doesn't support that message.

2. I understand that it is difficult to understand, it's all the same name. There are the PX4 stack and the ArduPilot stack. In the top of the stack we have the autopilot and they are different. In the bottom we have the operating system, NuttX, and they use the same.

alexblack, 2016-05-19:
1. Ok thanks.
2. Sorry your explanation isn't clear to me.  When building the firmware for the solo I see that PX4 firmware is needed, and ArduPilot is needed.  This seems to conflict with your statement that Solo does not use PX4.

alexblack, 2016-05-19:
I'm looking here:

https://github.com/3drobotics/PX4Firmware-solo/blob/5e69327474e2f954cdefddf11d0a64e654707440/src/modules/mavlink/mavlink_receiver.cpp#L702

OXINARF, 2016-05-19:
You are caring too much about the name. Let me try to make an analogy: both PX4 and ArduPilot are apps running on Windows. The thing is PX4 is hosted in the same repository as Windows - and that repository is called PX4Firmware.
ArduPilot is in a different repository, but it wants to use Windows so it has to link to the PX4Firmware repository.

Does that make sense?

alexblack, 2016-05-19:
Thats not really helping.  Can you just refer to real things?

1. I see that Solo uses Ardupilot: solo-ardupilot
2. I see that Solo uses PX4Nuttx (the OS that runs on Pixhawk?): PX4Nuttx

What then is solo-PX4Firmware used for?

OXINARF, 2016-05-19:
PX4Firmware is in the middle of ArduPilot and NuttX. It contains drivers for IMUs, for example.

alexblack, 2016-05-19:
Thanks, thats clear. So it sounds like the name PX4 refers to both the low level firmware needed by ArduPilot, AND, it refers to an autopilot that is a substitute for ArduPilot. 

Is that correct?

OXINARF, 2016-05-19:
Yes!

peterbarker, 2016-05-19:
On Wed, 18 May 2016, Alex Black wrote:

> Thanks, thats clear. So it sounds like the name PX4 refers to both the low level firmware needed by ArduPilot, AND, it refers to an autopilot that is a
> substitute for ArduPilot.
> 
> Is that correct?

Sometimes "PX4 Flight Core" and "PX4Firmware" are used to disambiguate.

OXINARF, 2016-05-19:
And I think they changed the "Flight Core" to PX4 Pro now? But they still have their "Flight Core" inside the "Firmware" repository, which makes it confusing for people.

closed 2016-06-05 by alexblack