3drobotics/solodevguide #311: How to connect Odroid to Pixhawk2?

pchatrath, 2016-05-23:
Hello,

I wish to create an ir tracker uav by connecting IR camera like pointgrey chameleon 3 ir camera to 3DR solo. I was thinking to connect a Odroid to Pixhawk 2 and make it the main brain which will take data from senor and pass commands to pixhawk using mavros. For that how do I connect odroid to Pixhawk?

Also for this application can i some how use the other processor available on 3DR solo. 

Thanks

hamishwillee, 2016-05-23:
@pchatrath I very much doubt 3DR will have time to provide more information than that already made available on http://dev.3dr.com/. 
This sort of question best asked on the [discussion boards](https://discuss.dronekit.io/c/solo-development).

That said, there is a very capable companion computer on the vehicle already. I imagine attaching an odroid as well doesn't make much sense.

pchatrath, 2016-05-24:
Thanks @hamishwillee I will ask the question at discussion board

jmachuca77, 2016-05-24:
Try this:

http://ardupilot.org/dev/docs/odroid-via-mavlink.html

Pixhawk2 is no different than a regular Pixhawk, just connect to the serial port on the accessory bay or on the gimbal connector instead of the telemetry port.

pchatrath, 2016-05-26:
Thanks @jmachuca77 . Will try it out

closed 2016-05-26 by pchatrath