3drobotics/solodevguide #313: Unable to capture video at 1920x1080

alexblack, 2016-05-28:
I've modified `libsolocam` to set the video capture resolution to 1920x1080, and set my GoPro to that same resolution, but, whenever I check the resolution after setting it, its still on 1280x720.  I also explored enumerating the supported formats, and the only resolution I get back is 1280x720, regardless of what resolution I have set the actual gopro to. 

Any ideas?

```
/// Grayscale = Y plane of YUV420
int solocam_set_format_1080p30_grayscale(struct solocam_ctx* ctx) {
  struct v4l2_streamparm streamparm;
  
  streamparm.type = V4L2_BUF_TYPE_VIDEO_CAPTURE;
  streamparm.parm.capture.capturemode = 4;
  streamparm.parm.capture.timeperframe.denominator = 30;
  streamparm.parm.capture.timeperframe.numerator = 1;
  
  if (-1 == xioctl(ctx->fd, VIDIOC_S_PARM, &streamparm)) {
    logerr("VIDIOC_S_PARM");
    return errno;
  }
 
  CLEAR(ctx->fmt);

  const int WIDTH = 1920;
  const int HEIGHT = 1080;

  ctx->fmt.type = V4L2_BUF_TYPE_VIDEO_CAPTURE;
  ctx->fmt.fmt.pix.width       = WIDTH;
  ctx->fmt.fmt.pix.height      = HEIGHT;
  ctx->fmt.fmt.pix.pixelformat = V4L2_PIX_FMT_YUV420;
  ctx->fmt.fmt.pix.field       = V4L2_FIELD_ANY;

  if (-1 == xioctl(ctx->fd, VIDIOC_S_FMT, &ctx->fmt)) {
    logerr("VIDIOC_S_FMT");
    return errno;
  }

  if (-1 == xioctl(ctx->fd, VIDIOC_G_FMT, &ctx->fmt)){
    logerr("VIDIOC_G_FMT");
    return errno;
  }

  if (ctx->fmt.fmt.pix.width != WIDTH || ctx->fmt.fmt.pix.height != HEIGHT) {
    fprintf(stdout, "Expected %dx%d, got %dx%d\n", WIDTH, HEIGHT, ctx->fmt.fmt.pix.width, ctx->fmt.fmt.pix.height);
    logerr("VIDIOC_G_FMT");  
    return -1;  
  }

  /* Buggy driver paranoia. */
  int min = ctx->fmt.fmt.pix.width * 2;
  if (ctx->fmt.fmt.pix.bytesperline < min)
          ctx->fmt.fmt.pix.bytesperline = min;
  min = ctx->fmt.fmt.pix.bytesperline * ctx->fmt.fmt.pix.height;
  if (ctx->fmt.fmt.pix.sizeimage < min)
          ctx->fmt.fmt.pix.sizeimage = min;
          
  return 0;
}
```

alexblack, 2016-05-29:
Can't seem to do this with `gst-launch` either:

```
root@3dr_solo:/opt/ddc# gst-launch mfw_v4lsrc num-buffers=1 capture-mode=5 !  jpegenc ! filesink location=sample.jpeg
MFW_GST_V4LSRC_PLUGIN 3.0.11 build on Mar  4 2016 02:10:26.
Setting pipeline to PAUSED ...
ERROR: Pipeline doesn't want to pause.
Setting pipeline to NULL ...
Freeing pipeline ...
```

ayushajain, 2016-07-19:
Has anyone figured out a way around this yet?