3drobotics/solodevguide #314: Removing background processes - CPU is being used

alexblack, 2016-05-31:
I'm processing images on the Solo, and realizing that performance is slower than it might be due to background processes using some of the CPU. For example, after killing some processes time to run `cv2.Canny` on an image dropped from 250ms to 150ms.

Can I kill these processes without any impact?

* `/usr/bin/main.py` - Looks like used for smart shot?
* `/usr/bin/telem_forwarder` - Looks like used for smart shot?
* `/usr/bin/pixrc` - when I killed this the drone made a noise and lights flashed... d'oh

peterbarker, 2016-05-31:
On Mon, 30 May 2016, Alex Black wrote:

> Can I kill these processes without any impact?

No.  They're there for a reason :-)  OTOH, you can probably do without 
some of the functionality.

>  *  /usr/bin/main.py - Looks like used for smart shot?

Not sure what this does. I do believe it is smart shots, 'though.

>  *  /usr/bin/telem_forwarder - Looks like used for smart shot?

telem_forwarder takes telemetry data from the PixHawk and gives it out to 
clients - most notably Artoo.

Kill this one if you don't care about being able to fly your Copter....

>  *  /usr/bin/pixrc - when I killed this the drone made a noise and lights flashed... d'oh

Likewise :-)

-- 
Peter Barker                          |   Programmer,Sysadmin,Geek.
pbarker@barker.dropbear.id.au	      |   You need a bigger hammer.
:: It's a hack! Expect underscores! - Nigel Williams

alexblack, 2016-05-31:
@peterbarker ok, so sounds like I *can* kill `/usr/bin/.main.py` then?  

What is Artoo?

alexblack, 2016-05-31:
Say my goal here is to fly the drone as bare bones as possible, is there anything else I might be able to do without?

hamishwillee, 2016-05-31:
Artoo is the internal name for the Solo Controller. The forwarder sends telemetry to the controller.

alexblack, 2016-06-05:
I've freed up a lot of background cpu by commenting out these lines in `/etc/inittab`:

```
#TOP:345:respawn:proc_top /log/3dr-top.log
#TEMP:345:respawn:log_temp
#DSYN:4:respawn:dataFlashMAVLink-to-artoo.py
#DFL:4:respawn:dataflash_logger
#API:4:respawn:run_shotmanager.sh
```

Then, before calling `dronekit.connect` you need to fire up Mav (which used to be done, along with shot manager, in `/usr/bin/run_shotmanager.sh`:

```
def startMav():
  """Requried if you're not running `/usr/bin/run_shot_manager.sh` on the Solo"""
  print "Starting up MAV..."
  from dronekit_solo import SoloVehicle
  from dronekit.mavlink import MAVConnection
  mav_vehicle = dronekit.connect("udpout:127.0.0.1:14560", wait_ready=False, vehicle_class=SoloVehicle, source_system=255, use_native=True, heartbeat_timeout=-1)
  out = MAVConnection("udpout:127.0.0.1:14550", source_system=254)
  mav_vehicle._handler.pipe(out)
  out.start()    
  print "Sleeping 5s to let MAV start..."
  # Not sure if this sleep is needed or not...
  time.sleep(5)
  print "Started MAV."
```