3drobotics/solodevguide #316: libsolocam.so Random Error freezes Solo

David-Proconsi, 2016-06-08:
I am using libsolocam.so without modifications in order to read frames from the camera but after a few seconds or minutes, it depends, the Solo freezes and start flashing red lights.
I tried it on the ground but i think that if it happens in flight it can be dangerous.

The error in the console is:

`python: solocam.c:300: solocam_read_frame: Assertion `buf.index < ctx->num_buffe                     rs' failed.
`

I can not understand exactly why it is produced and how to prevent it.
I think that the buffer is full, but i am not sure. I think it is produced in the line 329 of solocam.c
Anyone having the same problem? 
Do you know how to solve it?

alexblack, 2016-06-08:
Hi @David-Proconsi see my solution to https://github.com/3drobotics/solodevguide/issues/302, it might be the issue.   Though, while I've had similar trouble to you, eg only able to read a few frames before things go wrong, I haven't seen that specific error.

David-Proconsi, 2016-06-14:
I will try it as soon as possible. Thanks