3drobotics/solodevguide #320: solo-cli helloworld.py running

janthony, 2016-07-10:
When I do `solo script run helloworld.py`, it fails with a connection timeout everytime. Is the workaround to start a mavproxy manually?

peterbarker, 2016-07-12:
On Sat, 9 Jul 2016, Jerome Anthony wrote:

> When I do solo script run helloworld.py, it fails with a connection timeout
> everytime. Is the workaround to start a mavproxy manually?

No mavproxy should be required.

This is what I'm getting:

pbarker@bluebottle:~/rc/solo/solodevguide/examples/helloworld(fix-examples-webserver)$ 
solo script run helloworld.py
connecting to solo...
checking pip...
uploading script bundle...
./
./wheelhouse/
./wheelhouse/pymavlink-2.0.3-py2-none-any.whl
./wheelhouse/monotonic-0.6-cp27-none-any.whl
./wheelhouse/dronekit-2.7.0-py2-none-any.whl
./helloworld.py
./requirements.txt
./README.md
Not deleting env/bin
New python executable in env/bin/python
Installing setuptools, pip, wheel...done.
cp: can't stat '/usr/lib/python2.7/site-packages/greenlet*': No such file 
or directory
Ignoring indexes: https://pypi.python.org/simple
Processing ./wheelhouse/dronekit-2.7.0-py2-none-any.whl
Processing ./wheelhouse/monotonic-0.6-cp27-none-any.whl
Processing ./wheelhouse/pymavlink-2.0.3-py2-none-any.whl
Installing collected packages: pymavlink, monotonic, dronekit
Successfully installed dronekit-2.7.0 monotonic-0.6 pymavlink-2.0.3
running script...

>>> APM:Copter solo-1.2.20 (b5013474)
>>> PX4: 5e693274 NuttX: d48fa307
>>> Frame: QUAD
>>> PX4v2 004A002F 33345119 32383433
>>> PreArm: Need 3D Fix
>>> PreArm: Need 3D Fix
ERROR LOADING MAVNATIVE - falling back to python implementation
Using MAVLink 1.0
Connecting to udpin:0.0.0.0:14550...
Vehicle state:
  Global Location: 
LocationGlobal:lat=-35.ish,lon=149.ish,alt=500ish
  Global Location (relative altitude): 
LocationGlobalRelative:lat=-35.ish,lon=149.ish,alt=0.94
  Local Location: LocationLocal:north=None,east=None,down=None
  Attitude: 
Attitude:pitch=-0.00305004091933,yaw=2.54118466377,roll=-0.00677156820893
  Velocity: [0.040000000000000001, 0.0, 0.02]
  Battery: Battery:voltage=15.036,current=0.68,level=24
  Last Heartbeat: 0.183725333
  Heading: 145
  Groundspeed: 0.329999983311
  Airspeed: 0.329999983311
  Mode: LOITER
  Is Armable?: False
  Armed: False
Done.
Exception in thread Thread-2 (most likely raised during interpreter 
shutdown):
Traceback (most recent call last):
   File "/usr/lib/python2.7/threading.py", line 551, in __bootstrap_inner
   File "/usr/lib/python2.7/threading.py", line 504, in run
   File 
"/log/solo-script/env/lib/python2.7/site-packages/dronekit/mavlink.py", 
line 166, in mavlink_thread_out
<type 'exceptions.AttributeError'>: 'NoneType' object has no attribute 
'error'
Exception in thread Thread-1 (most likely raised during interpreter 
shutdown):
Traceback (most recent call last):
   File "/usr/lib/python2.7/threading.py", line 551, in __bootstrap_inner
   File "/usr/lib/python2.7/threading.py", line 504, in run
   File 
"/log/solo-script/env/lib/python2.7/site-packages/dronekit/mavlink.py", 
line 206, in mavlink_thread_in
<type 'exceptions.AttributeError'>: 'NoneType' object has no attribute 
'error'
pbarker@bluebottle:~/rc/solo/solodevguide/examples/helloworld(fix-examples-webserver)$

janthony, 2016-07-12:
@peterbarker Thanks for checking. Your output is exactly like something I hope to see. Are you running with the latest solo and controller firmware updates?

franknitty69, 2016-07-12:
FYI this and #645 and the same issue

peterbarker, 2016-07-12:
bOn Mon, 11 Jul 2016, Jerome Anthony wrote:

> @peterbarker Thanks for checking. Your output is exactly like something I
> hope to see. Are you running with the latest solo and controller firmware
> updates?

Yes.  Before the update it all worked as expected.

After the update to the latest versions I had to specify the modified 
connection string.

Amusingly, it seems the Solo app on my phone no longer sees Solo - that 
can't be a good thing....